petercorke/robotics-toolbox-python #296: fkine_all not working for UR5

rhys-newbury, 2022-02-28:
**Describe the bug**
fkine_all not working for the UR5 robot. 

**Version information**
Installed from PyPi (using Conda) v0.11.0

**To Reproduce**
`
panda = rtb.models.Panda()
panda.q = panda.qr

print(panda.fkine_all(panda.q))

ur5 = rtb.models.UR5()
ur5.q = ur5.qr

print(ur5.fkine_all(ur5.q))
`

**Expected behavior**
The UR5 should calculate the forward kinematics of all of the joints, however, the code crashes.
 Output:
`
Traceback (most recent call last):
  File "C:\Users\newbu\test\ur5_isolated.py", line 12, in <module>
    print(ur5.fkine_all(ur5.q))
  File "D:\conda\envs\empty\lib\site-packages\roboticstoolbox\robot\ERobot.py", line 683, in fkine_all
    recurse(linkframes, Tbase, q, self.links[0])
  File "D:\conda\envs\empty\lib\site-packages\roboticstoolbox\robot\ERobot.py", line 665, in recurse
    Tall[link.number] = T
  File "D:\conda\envs\empty\lib\site-packages\spatialmath\baseposelist.py", line 328, in __setitem__
    self.data[i] = value.A
IndexError: list assignment index out of range
`

**Environment (please complete the following information):**
Windows 10

rhys-newbury, 2022-02-28:
If I remove the extra frames (world frame, extra end-effector frame, and rotated base frame) in the UR5 definition, the code now works. But it might be better to come up with a less hacky fix.

jhavl, 2022-04-07:
Thanks for finding that Rhys, this is fixed on the future branch and will be included in the next release.

commit 8a63a120 (2022-04-07): Fix #296 link count was incorrect for erobots
closed 2022-04-07 by jhavl