petercorke/robotics-toolbox-python #306: Build is failing of gh-actions

tdamsma, 2022-04-15:
I noticed the latest master branch builds are failing. Would you accept PR's addressing this? Also I found some missing dependencies on a plain installation. Would be happy to contribute something back to this package

walchko, 2022-04-16:
python: 3.9.10
os: mac 12.2.1
robotics-toolbox: 0.11.0

Yes, tried to load this model and it failed:

```python
al5d = rtb.models.DH.AL5D()
print(al5d)
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Input In [18], in <cell line: 1>()
----> 1 al5d = rtb.models.DH.AL5D()
      2 print(al5d)

AttributeError: module 'roboticstoolbox.models.DH' has no attribute 'AL5D'
```

When I copied and pasted the code from AL5D.py into Jupyter and tried it again, I got this error which  is more than missing AL5D dependency:

```python
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Input In [14], in <cell line: 1>()
----> 1 al5d = AL5D(symbolic=False)
      2 print(al5d)

Input In [13], in AL5D.__init__(self, symbolic)
     69 links = []
     71 for j in range(4):
---> 72     link = RevoluteMDH(
     73         d=d[j],
     74         a=a[j],
     75         alpha=alpha[j],
     76         offset=offset[j],
     77         r=center_of_mass[j],
     78         I=moments_of_inertia[j],
     79         G=1,
     80         B=0,
     81         Tc=[0,0],
     82         qlim=joint_limits[j]
     83     )
     84     links.append(link)
     86 tool=SE3(0.07719,0,0)

File ~/robottoolkit/lib/python3.9/site-packages/roboticstoolbox/robot/DHLink.py:741, in RevoluteMDH.__init__(self, d, a, alpha, offset, qlim, flip, **kwargs)
    738 sigma = 0
    739 mdh = True
--> 741 super().__init__(
    742     d=d,
    743     alpha=alpha,
    744     theta=theta,
    745     a=a,
    746     sigma=sigma,
    747     mdh=mdh,
    748     offset=offset,
    749     qlim=qlim,
    750     flip=flip,
    751     **kwargs,
    752 )

File ~/robottoolkit/lib/python3.9/site-packages/roboticstoolbox/robot/DHLink.py:135, in DHLink.__init__(self, d, alpha, theta, a, sigma, mdh, offset, **kwargs)
    127 def __init__(
    128     self, d=0.0, alpha=0.0, theta=0.0, a=0.0, sigma=0, mdh=False, offset=0, **kwargs
    129 ):
   (...)
    133     #  probably should enforce keyword arguments, easy to make an
    134     #    error with positional args
--> 135     super().__init__(**kwargs)
    137     # Kinematic parameters
    138     self.sigma = sigma

File ~/robottoolkit/lib/python3.9/site-packages/roboticstoolbox/robot/Link.py:119, in Link.__init__(self, name, qlim, flip, m, r, I, Jm, B, Tc, G, mesh, geometry, collision, **kwargs)
    117 # Joint angle parameters
    118 self.flip = flip
--> 119 self.qlim = qlim
    121 # Link geometry
    122 self.geometry = geometry

File ~/robottoolkit/lib/python3.9/site-packages/roboticstoolbox/robot/Link.py:485, in Link.qlim(self, qlim_new)
    483     self._qlim = None
    484 else:
--> 485     self._qlim = getvector(qlim_new, 2)

File ~/robottoolkit/lib/python3.9/site-packages/spatialmath/base/argcheck.py:331, in getvector(v, dim, out, dtype)
    328     dt = None
    330 if dim is not None and v and len(v) != dim:
--> 331     raise ValueError("incorrect vector length")
    332 if out == "sequence":
    333     return v

ValueError: incorrect vector length
```

jhavl, 2022-07-04:
Hey @tdamsma, thanks for your interest! We are always happy to get pull requets for new features and bug features. Ideally keep each pull request small, self contained and within the style of the package

closed 2022-07-04 by jhavl