petercorke/robotics-toolbox-python #308: Reference developer home folder

walchko, 2022-04-16:
Suggest fixing this, there is no point to deploy code that references a developers home folder. It also happens in another place in the code.

https://github.com/petercorke/robotics-toolbox-python/blob/176f21cd8e7611c74b3346ccfd5fce470f6ec487/roboticstoolbox/examples/mobile.py#L13

petercorke, 2023-04-21:
commented out for now, but really need to develop some example scripts.  Some of these are covered in the notebooks [here](https://github.com/petercorke/RVC3-python/tree/main/notebooks).  Data files are are now shipped with the toolbox and accessible in an installation independent manner.

petercorke, 2026-09-19:
Thank you for your interest in RTB! The lines referencing a developer's local home folder have been commented out since 2023, and data files (`house.mat`, `map1.mat`, etc.) now ship with the package and are located via `rtb_path_to_datafile()` rather than a hardcoded path. Closing as resolved.

closed 2026-09-19 by petercorke