petercorke/robotics-toolbox-python #322: Example on https://jhavl.github.io/holistic/ not working

chisarie, 2022-06-28:
Dear Jesse and Peter, 

thank you very much for the work you are doing with this toolbox, I just came across it and it looks amazing!

I was trying to run the code from https://jhavl.github.io/holistic/ (the omnidirectional base one), but unfortunately I run into some errors. Some errors I run into:
 
```
Traceback (most recent call last):
  File "/home/chisari/git/manipulation_examples/roboticstoolbox/demo.py", line 111, in <module>
    arrived, frankie.qd = step_robot(frankie, wTep.A)
  File "/home/chisari/git/manipulation_examples/roboticstoolbox/demo.py", line 12, in step_robot
    wTe = r.fkine(r.q, fast=True)
TypeError: fkine() got an unexpected keyword argument 'fast'
```
```
Traceback (most recent call last):
  File "/home/chisari/git/manipulation_examples/roboticstoolbox/demo.py", line 111, in <module>
    arrived, frankie.qd = step_robot(frankie, wTep.A)
  File "/home/chisari/git/manipulation_examples/roboticstoolbox/demo.py", line 63, in step_robot
    θε = math.atan2(bTe[1, -1], bTe[0, -1])
  File "/home/chisari/miniconda3/envs/mujoco_env/lib/python3.9/site-packages/spatialmath/baseposelist.py", line 322, in __getitem__
    ret = self.__class__(self.data[i], check=False)
TypeError: list indices must be integers or slices, not tuple
```

<details>
  <summary>For completeness, I paste here the whole script: (Click to expand)</summary>
  
  ## Heading
  1. A numbered
  2. list
     * With some
     * Sub bullets
</details>

Is it possible that the code on the website refers to an older version of the toolbox? If yes, it would be great if you could provide an updated version of it, I really looking forward to try it out!

**Version information**

Did you install from PyPI or GitHub? I tried both, same effect
If PyPI what version number? 1.0.1
If GitHub what commit hash? 1.0.1

**Environment (please complete the following information):**

 - Your OS (MacOS, Linux, Windows). Ubuntu 20
- Your Python version. 3.9.12

chisarie, 2022-06-30:
@jhavl

jhavl, 2022-07-04:
Hey @chisarie, thanks for finding this. I have updated the examples on https://jhavl.github.io/holistic/ which will work with the latest toolbox version. These examples are also available in the examples directory on this repo: [holistic_mm_non_holonomic.py](https://github.com/petercorke/robotics-toolbox-python/blob/master/roboticstoolbox/examples/holistic_mm_non_holonomic.py) and [holistic_mm_holonomic.py](https://github.com/petercorke/robotics-toolbox-python/blob/master/roboticstoolbox/examples/holistic_mm_omni.py)