petercorke/robotics-toolbox-python #323: module 'sympy.core.core' has no attribute 'symbol'

LorenzoRamirez, 2022-07-04:
**Check here first**

[Common issues](https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues)

**Describe the bug**
A clear and concise description of what the bug is.
I down load and install  roboticstoolbox-python (1.0.1)

**Version information**
 roboticstoolbox-python (1.0.1)
Did you install from PyPI or GitHub?
If PyPI what version number?
If GitHub what commit hash?
I install using pip

Robotics Toolbox depends heavily on two other packages: Swift (3D graphics) and SpatialMath toolbox (underpinning maths utilities).  If you think your issue is related to these, then please answer the questions above for them.

**To Reproduce**
Steps to reproduce the behavior:
1. The shortest, complete, Python script that exhibits the bug.
2. The script output, including error messages.
I  try in Jupyter Notebook the following comnad:

import roboticstoolbox as rtb

and I got the follwoing error:
_____________
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/var/folders/mm/6h83jcs57_qb997r_m_wyzkc0000gq/T/ipykernel_69066/875870540.py in <module>
----> 1 import roboticstoolbox as rtb

/Users/Shared/my_project_folder/anaconda3/envs/py3k/anaconda3_9-1-2021/anaconda3/lib/python3.8/site-packages/roboticstoolbox/__init__.py in <module>
      1 from roboticstoolbox.tools import *
      2 
----> 3 from roboticstoolbox.robot import *
      4 
      5 from roboticstoolbox.mobile import *

/Users/Shared/my_project_folder/anaconda3/envs/py3k/anaconda3_9-1-2021/anaconda3/lib/python3.8/site-packages/roboticstoolbox/robot/__init__.py in <module>
----> 1 from roboticstoolbox.robot.Robot import Robot
      2 from roboticstoolbox.robot.Link import Link, Link2
      3 from roboticstoolbox.robot.DHRobot import SerialLink, DHRobot
      4 from roboticstoolbox.robot.DHLink import (
      5     DHLink,

/Users/Shared/my_project_folder/anaconda3/envs/py3k/anaconda3_9-1-2021/anaconda3/lib/python3.8/site-packages/roboticstoolbox/robot/Robot.py in <module>
     17 from roboticstoolbox.backends.PyPlot.EllipsePlot import EllipsePlot
     18 from roboticstoolbox.robot.Dynamics import DynamicsMixin
---> 19 from roboticstoolbox.robot.ETS import ETS
     20 from roboticstoolbox.robot.IK import IKMixin
     21 from typing import Optional, Union, overload, Dict, Tuple

/Users/Shared/my_project_folder/anaconda3/envs/py3k/anaconda3_9-1-2021/anaconda3/lib/python3.8/site-packages/roboticstoolbox/robot/ETS.py in <module>
     40 from copy import deepcopy
     41 from roboticstoolbox import rtb_get_param
---> 42 from roboticstoolbox.robot.ET import ET, ET2
     43 from spatialmath.base import getvector
     44 from spatialmath import SE3

/Users/Shared/my_project_folder/anaconda3/envs/py3k/anaconda3_9-1-2021/anaconda3/lib/python3.8/site-packages/roboticstoolbox/robot/ET.py in <module>
     30 
     31     # Sym = sympy.Expr
---> 32     Sym = sympy.core.symbol.Symbol
     33 
     34 except ImportError:  # pragma: no cover

AttributeError: module 'sympy.core.core' has no attribute 'symbol'

________________

**Expected behavior**
A clear and concise description of what you expected to happen.
I was expecting no error when executing the command:

import roboticstoolbox as rtb

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Your OS (MacOS, Linux, Windows).
- Your Python version.
- 
I am on MAC  High Sierra  (10.13.6)
Python  ipython                       8.3.0
sympy                         1.5.1
**Additional context**
Add any other context about the problem here.
This is the first time I tried to use roboticstoolsbox-python in Jupyter Notebook.
It seems that the issue could be with Sympy but I do not know what to do in order to solve the issue.  
I search for a similar issue but I did not find any reported issue.
Any help will be greatly appreciated.
Thanks

jhavl, 2022-07-04:
Hey @LorenzoRamirez, could you please share the output of
```
pip list | grep sympy
```

LorenzoRamirez, 2022-07-05:
Jesse
Here it is the requested output: -----------------

	(base) Macintosh-6:~ JCarrington$ pip list | grep sympy

	sympy  1.5.1

	WARNING: You are using pip version 20.3.4; however, version 22.1.2 is
available.

	You should consider upgrading via the
'/Users/Shared/my_project_folder/anaconda3/envs/py3k/anaconda3_9-1-2021/anaconda3/bin/python
-m pip install --upgrade pip' command.-------------------
Please let me know if additional information is required and thanks
for the help!
Lorenzo Ramirez

	-----------------------------------------From: "Jesse Haviland" 
To: "petercorke/robotics-toolbox-python"
Cc: "Lorenzo", "Mention"
Sent: Monday July 4 2022 2:20:27AM
Subject: Re: [petercorke/robotics-toolbox-python] module
'sympy.core.core' has no attribute 'symbol' (Issue #323)

	Hey @LorenzoRamirez [1], could you please share the output of 

Links:
------
[1] https://github.com/LorenzoRamirez
[2]
https://github.com/petercorke/robotics-toolbox-python/issues/323#issuecomment-1173398405
[3]
https://github.com/

JefferyChiang, 2022-07-07:
upgrade sympy  to 1.10.1 solve the problem

LorenzoRamirez, 2022-07-16:
JefferyThanks for the email.
As recommended I upgrade sympy and the error went away.
FYI . I am dealing with another error that I will report in the
sissue:ImportError: cannot import name 'UGraph' from 'pgraph'
I tried few things but I can ot resolve the issue.
Take care 
Lorenzo Ramirez

	-----------------------------------------From: "JefferyChiang" 
To: "petercorke/robotics-toolbox-python"
Cc: "Lorenzo", "Mention"
Sent: Wednesday July 6 2022 11:30:40PM
Subject: Re: [petercorke/robotics-toolbox-python] module
'sympy.core.core' has no attribute 'symbol' (Issue #323)

	upgrade sympy to 1.10.1 solve the problem 

	—
[2].

Links:
------
[1]
https://github.com/petercorke/robotics-toolbox-python/issues/323#issuecomment-1177012179
[2]

petercorke, 2023-04-21:
closing the sympy issue now.  works now with no simpy installed or latest version 1.11 installed.

closed 2023-04-21 by petercorke