petercorke/robotics-toolbox-python #324: Installing both numba and roboticstoolbox breaks roboticstoolbox

sdiebolt, 2022-07-06:
**Describe the bug**
Installing `roboticstoolbox-python` in an environment with an existing numba install leads to the following error when trying to import roboticstoolbox:

> RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xf
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/sdiebolt/.cache/pypoetry/virtualenvs/pyfus-ae9x7n-S-py3.9/lib/python3.9/site-packages/roboticstoolbox/__init__.py", line 3, in <module>
    from roboticstoolbox.robot import *
  File "/home/sdiebolt/.cache/pypoetry/virtualenvs/pyfus-ae9x7n-S-py3.9/lib/python3.9/site-packages/roboticstoolbox/robot/__init__.py", line 1, in <module>
    from roboticstoolbox.robot.Robot import Robot
  File "/home/sdiebolt/.cache/pypoetry/virtualenvs/pyfus-ae9x7n-S-py3.9/lib/python3.9/site-packages/roboticstoolbox/robot/Robot.py", line 19, in <module>
    from roboticstoolbox.robot.ETS import ETS
  File "/home/sdiebolt/.cache/pypoetry/virtualenvs/pyfus-ae9x7n-S-py3.9/lib/python3.9/site-packages/roboticstoolbox/robot/ETS.py", line 43, in <module>
    from fknm import (
ImportError: numpy.core.multiarray failed to import

**Version information**
`roboticstoolbox-python` was installed from PyPi, version 1.0.2. Numba was also installed from PyPi, version 0.55.2.

**To Reproduce**
1. Create a new Python 3.10 environment.
2. Install numba using `pip install numba`
3. Install roboticstoolbox using `pip install roboticstoolbox-python`
4. Launch a python interpreter and run `import roboticstoolbox`.

**Expected behavior**
roboticstoolbox should be imported without error.

**Environment (please complete the following information):**
 - Ubuntu 22.04
- Python 3.10

**Additional context**
I think the problem might come from the fact that numba uses numpy<1.23.0, while roboticstoolbox has no max numpy version. From the error, it seems like roboticstoolbox was built using a different numpy version than the one installed. I might be wrong though, I'm not that familiar wit Python module building.

jhavl, 2022-12-09:
Hey @sdiebolt, I believe we have fixed this with the latest release. We now build our wheels with very early versions of numpy which should remove the above error.

closed 2022-12-09 by jhavl

sdiebolt, 2022-12-16:
It is indeed fixed, thanks a lot!