petercorke/robotics-toolbox-python #338: pyplot = rtb.backends.PyPlot()  # create a PyPlot backend TypeError: 'module' object is not callable

River-mao, 2022-10-13:
Hello! When I try the backends example code in https://petercorke.github.io/robotics-toolbox-python/arm_backend_pyplot.html, 
'''
import roboticstoolbox as rtb

robot = rtb.models.DH.Panda()  # create a robot

pyplot = rtb.backends.PyPlot()  # create a PyPlot backend
pyplot.add(robot)              # add the robot to the backend
robot.q = robot.qz             # set the robot configuration
pyplot.step()                  # update the backend and graphical view
'''
I encountered the following error:
'''
pyplot = rtb.backends.PyPlot()  # create a PyPlot backend
TypeError: 'module' object is not callable
'''
I need some help. Thanks a lot.

lulu110997, 2023-11-13:
I'm running into the same error when calling rtb.backends.PyPlot(). Did you find a solution?

lulu110997, 2023-11-13:
This worked for me

`import roboticstoolbox as rtb

robot = rtb.models.DH.Panda()  # create a robot

pyplot = rtb.backends.PyPlot.PyPlot()  # create a PyPlot backend
pyplot.launch()
pyplot.add(robot)              # add the robot to the backend
robot.q = robot.qz             # set the robot configuration
pyplot.step()                  # update the backend and graphical view
pyplot.hold()`

Alternatively, if you want to animate a robot, you can just use the figure returned by the robot.plot() method. For example

`from spatialmath import SE3, SE2

import numpy as np
import math
import roboticstoolbox as rtb

robot = rtb.DHRobot(
    [
        rtb.RevoluteDH(a=1),
        rtb.RevoluteDH(a=1)
    ], name="2D_robot")
print(robot)

t = np.arange(0, 2, 0.1)  # time
T0 = SE3(-1, 0.5, 0)  # initial pose
T1 = SE3(1, 1.5, 0)  # final pose
Ts = rtb.tools.trajectory.ctraj(T0, T1, t)

robot_fig = robot.plot([0, 0])
robot_fig.ax.set(xlim=(-1.5, 2), ylim=(-1.5, 2))

sol = robot.ikine_LM(Ts)

for q in sol.q:
    robot.q = q  # set the robot configuration
    robot_fig.step()
robot_fig.hold()`

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking well over four years to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one: turns out the docs example you were following was just wrong, and it's been wrong the whole time — `roboticstoolbox.backends.PyPlot()` was never a valid way to construct the backend (even the workaround posted in this thread, `rtb.backends.PyPlot.PyPlot()`, doesn't work either — I checked directly). The correct form is `from roboticstoolbox.backends.PyPlot import PyPlot; pyplot = PyPlot()`. Fixed the actual docs in #641, verified the corrected example really runs and draws the robot. Thank you for flagging this — it's been broken in the docs for a long time.

closed 2026-08-25 by petercorke