petercorke/robotics-toolbox-python #343: `ETS_fkine` outputs wrong `SE3` when sequence of `q` is passed

FelipeCybis, 2022-11-29:
**Check here first**
[Common issues](https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues)
No related issue was found here.

**Describe the bug**
Forward kinematics in custom ERobot in version > 1.0.0 now is forced (with `try / except`) to pass through the Cpp function `ETS_fkine`
But `ETS_fkine` does not handle the `q` parameter as its pythonic `fkine` does in versions < 1.0 used to and outputs wrong results when sequence of `q` is passed.

For example, creating this simple robot in `v1.0.2`: 
```
import roboticstoolbox as rtb
import numpy as np

robot = rtb.ERobot(
    [
        rtb.Link(rtb.ET.tz()),
        rtb.Link(rtb.ET.tx())
    ]
)
traj = np.array([[0,0],[0,1]])
print(robot.fkine(q=traj))
```
Gives us the wrong fkine:
```
0:
   1         0         1         0         
   0         1         0         1         
   0         0         0         1         
   0         0         0         0         
1:
   0         0         0         0         
   0         0         0         0         
   1         0         1         0         
   0         1         0         1  
```
Following the docstring, that states "**Trajectory operation: If ``q`` has multiple rows (mxn), it is considered a trajectory and the result is an ``SE3`` instance with ``m`` values.**", we would expect two SE(3) as a response for the two translations in `traj`, the first with `z=0, x=0` and the second with `z=0, x=1`, or:
```
0:
   1         0         0         0         
   0         1         0         0         
   0         0         1         0         
   0         0         0         1         
1:
   1         0         0         1         
   0         1         0         0         
   0         0         1         0         
   0         0         0         1      
```
The above result is indeed what we get in `v0.11.0` with fkine argument `fast=False` as well as by cloning the bleeding-edge version from commit 07ad1259338d6dfd41af6e5e50973e315284a912 and putting back on the `fast` argument so it can skip the `ETS_fkine` function (which I guess is the fastest workaround for now)

**Version information**
`v1.0.2` and `v0.11.0` installed from PyPI
commit 07ad1259338d6dfd41af6e5e50973e315284a912 from GitHub

**Environment (please complete the following information):**
 - Windows 11 10.0.22621 Build 22621
 - Conda version 22.9.0
 - Python 3.10.8
 Env created with
```
conda create -n rtb python=3.10.8
pip install roboticstoolbox-python
```

jhavl, 2023-04-19:
Hey @FelipeCybis, thanks for finding that, it was a tricky one in the end with column-major ordering with 3D arrays. It's been fixed and will be included in the next release

closed 2023-04-19 by jhavl

FelipeCybis, 2023-04-19:
Awesome, @jhavl! Thanks :)

commit 083cdc43 (2023-04-19): Fix #343 array ordering causing fkine traj error