petercorke/robotics-toolbox-python #345: NonCommercial Licensing concern

hyandell, 2023-01-19:
Hi Jesse and Peter,

In digging into this repository, I noticed that one of the files contains CC-NC-SA licensing. The file is:

    https://github.com/petercorke/robotics-toolbox-python/blob/master/roboticstoolbox/mobile/PoseGraph.py

It looks like the code was introduced in the initial commit:

    https://github.com/petercorke/robotics-toolbox-python/commit/c1a3267170dc98a781a8b2184270f49d775f63d5

This seems like it is at odds with the open source licensing of the project as a whole, so I wanted to flag it for your review.

borisclemencon, 2023-02-28:
Do we have any news for this issue? RTB is blocked for usage by our legal team because of this small piece of code.

jhavl, 2023-03-13:
Thanks for bringing this to our attention. We have gotten in touch with the owner of the referenced code and we have permission to swap the licence of concern with a simple reference. This has been fixed on the future branch and will be incorporated in the next release of the toolbox in early April.

closed 2023-03-13 by jhavl