petercorke/robotics-toolbox-python #346: Can robotics toolbox for python model a robotic manipulator like the ABB IRB 460?

freireguilherme, 2023-01-26:
I am currently doing a project in which I would like to investigate the kinematics and dynamics of the ABB IRB 460 model. Can the RTP model this type of robotic arm?
[image]

jhavl, 2023-04-19:
Hey @freireguilherme, yes this robot can be modelled using the toolbox. It can be added as a DH model, and ETS model, or as a URDF. After finding the kinematic details you can create one of these model types using the toolbox.

closed 2023-04-19 by jhavl

freireguilherme, 2023-05-20:
Hey @jhavl, thank you for the response!
Ok, I'm trying to add this model as a DH model. I have now the DH parameters. But how can I describe the passive joins and links? I mean, how to put the passive joins and links conected with the active joins and links?