petercorke/robotics-toolbox-python #347: Issue for custom model plot

gaolongsen, 2023-02-02:
Hello everyone, I tried to import my customed robot arm. Now I can import the model and the information of link and joint and all other information shows correctly. I can also follow simple examples to use the related kinetic and dynamic functions to get the right results. However, I meet two problems:
1. When I try to use the plot function as the simple in the main page, I can not see the robot move in the gif file, not only for my customed robot, but also for the other robot from the lb such as panda, ur5, and so on. 
2. When I try to use **Swift simulator** as the example shown for the Franka robot, the page in my browser shows that "_**An unexpected error has occurred.**_". I think my URDF file and related directory should be fine.
The above two problem makes me so confused and I hopes someone can support useful suggestions. Thanks for everyone's time!

jhavl, 2023-04-19:
Hey @gaolongsen, what version of the toolbox and swift are you using and can you also please share the model with us?

petercorke, 2026-08-25:
Sorry for the very long delay getting back to you. This one never got the extra detail asked for (toolbox/swift version, model file) needed to actually investigate, and a lot has changed since — Swift went through a full rewrite (2.0.0) and RTB's own plotting backends have had several fixes this session (see #338, #405, #480, #481, #487, #505, #509 if any look related to what you were hitting). Closing for now since there's nothing actionable left in the thread — please file a fresh issue with a minimal repro if you're still seeing this on the current release, happy to take another look.

closed 2026-08-25 by petercorke