petercorke/robotics-toolbox-python #350: KUKA iiwa LBR robot jacobian broken

JKBehrens, 2023-02-09:
Dear Peter, dear Jesse,

I am trying to use the toolbox for our iiwa robots, but there seems to be an issue with the models in the rtb data package. It would be great if you could look at the report below.

Thank you and all the best,
Jan

**Describe the bug**
The jacobian for the KUKA LBR robot is incorrect. The following works for Panda (i.e., the eef moves slowly in x direction), but not for the iiwa LBR. The LBR moves wildly around.

```
 v = np.array([0.05, 0, 0, 0, 0, 0])
 panda.qd = np.linalg.pinv(panda.jacobe(panda.q)) @ v
```

**Version information**

I created a new conda environment with Python 3.9 and installed the toolbox via `pip install roboticstoolbox-python[collision]`

```
roboticstoolbox-python    1.0.3                    pypi_0    pypi
rtb-data                  1.0.0                    pypi_0    pypi
spatialgeometry           1.0.3                    pypi_0    pypi
spatialmath-python        1.0.5                    pypi_0    pypi
```

**To Reproduce**
Steps to reproduce the behavior:
1. The shortest, complete, Python script that exhibits the bug.
```
import swift
import roboticstoolbox as rp
import numpy as np

env = swift.Swift()
env.launch(realtime=True)

# works with panda
# panda = rp.models.Panda()
# doesn't work with LBR
panda = rp.models.LBR()

panda.q = panda.qr

env.add(panda)

dt = 0.05

for _ in range(100):
    v = np.array([0.05, 0, 0, 0, 0, 0])
    panda.qd = np.linalg.pinv(panda.jacobe(panda.q)) @ v
    env.step(dt)
```

See the screenshots below. The LBR moves up and backwards. The Panda moves forward as expected.

**Expected behavior**
Both robots should move slowly forward.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
- Python 3.9

jhavl, 2023-04-19:
Hey @JKBehrens, thanks for bringing this to our attention. I've done some quick debugging and this seems to be the same issue as #361. This should be fixed once that one is fixed.

JKBehrens, 2023-04-19:
Great! I will try it.