petercorke/robotics-toolbox-python #356: vscode.dev import method with micropip

randomlogic78, 2023-03-10:
**Describe the bug**
There does not seem to be a path to import the package in pyolite/pyodide in vscode.dev in the browser or the webapp. This is particularly useful for students that must use a chromebook without the linux subsystem.

**Version information**

N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Open vscode.dev in chrome
2. install the Jupyter PowerToys extension from microsoft
3. Select the pyolite kernal
4. Run the following:

import micropip
await micropip.install("roboticstoolbox-python")

**Expected behavior**
Micropip loads the package into the pyolite kernel

**Screenshots**
N/A

**Environment (please complete the following information):**
 - Your OS MacOS, ChromeOS
- Your Python version: Pyolite 3.9.5

**Additional context**
N/A

jhavl, 2023-04-19:
Hey @randomlogic78, we're are excited about Pyodide and WASM but there is significant work needed before we will be able to support it. This is something we are investigating but not on the short term horizon. Are you working in this space or would you be interested in helping out here?

jhavl, 2023-04-28:
Closing this, reopen in Discussions if you would like to continue the conversation

closed 2023-04-28 by jhavl