petercorke/robotics-toolbox-python #361: flipped revolute joints causing confusing Jacobian for optimization

MiaoDragon, 2023-04-01:
Hi developers,

I'd like to first express my appreciation for this toolbox. It's very convenient and good for understanding the kinematics and dynamics of the robot. I understand that the toolbox is still in the development phase, and I hope it can be improved in the future to be more robust.
**Issue**
One issue I experienced when using the toolbox is that it didn't handle the generalized revolute joint axis. I understand that any such revolute joint axis can be converted using one extra dummy layer below it to transform the rotational axis. However, this would require a modification of the robot description files to match the format of Rx, Ry or Rz joint axis.
One specific issue caused by this is that when the joint axis is flipped, i.e. having a negative axis such as (0,0,-1), the toolbox will treat it as rotating at negative angles around (0,0,1). This causes a problem in the Jacobian computation and unmatched behaviors with other robotic simulators. Specifically, I have tested with Mujoco, and found that the jacobian values are negative of those in Mujoco for the flipped axis.
This could cause problems when we use this solver to compute Jacobian for custom optimization problems. For example, when I'm using Mujoco to simulate the configuration state, the Jacobian computed by rtb will give me the wrong direction to update the state.
**Suggestions**
One suggestion is to support a generic revolute joint axis. As described in the Jacobian tutorial, there is
$$\frac{\partial R(\theta)}{\partial\theta}=[\hat{\omega}]_\times R(\theta)$$
so it shouldn't be hard to implement.
Thank you!

miguelriemoliveira, 2023-04-03:
Hi,

I would also like to thank the authors for such a fantastic toolbox ... and in python!!!

I can confirm this problem. I was trying to implement the null space projection example, which uses the Jacobian computation, in a robot with negative joint values (it showed when the robot was printed) and it was not working. 

Setting the joints to positive works fine.

jhavl, 2023-04-19:
Hey @MiaoDragon, can you please give us the robot model for which this happens, along with a set of joint angles and what you believe the Jacobian should be at those joint values (from Mujoco or other environment)?

jhavl, 2023-04-19:
This has been fixed and will be included in the next release

commit 784ffbc2 (2023-04-19): numerical test for fix #361
commit f0ec2f8d (2023-04-19): fix #361 flipped joint causing Jacobian error
closed 2023-04-19 by jhavl