petercorke/robotics-toolbox-python #364: fkine start parameter bug

chisarie, 2023-04-25:
Hi, 

First of all, thank you very much for the effort in maintaining this very nice repo!

**Describe the bug**
It seems to me that there is a bug in how the `start` parameter of the `fkine` function works. It seems that it always takes the parent of the provided link, instead of the provided link itself.

**To Reproduce**

```python
import roboticstoolbox as rtb

robot = rtb.models.Panda()
print(robot)
transform = robot.fkine(robot.qz, start="panda_link1", end="panda_link2")
print(transform)
```

The transform I get has a translation in z-direction of 0.333, even though between link1 and link2 there should be no translation

**Version information**

Did you install from PyPI or GitHub? -> From PypI
If PyPI what version number? -> 1.0.3

**Environment (please complete the following information):**
 - Your OS: Ubuntu 20.04
- Your Python version: 3.9.16

jhavl, 2023-04-28:
Hi @chisarie, I believe this is working as intended. The forward kinematics is calculating the transform from the start of panda_link1 to the end of panda_link2. 

When we print the ETS from link1 to link2 we get 

```python
>>> print(robot.ets(start="panda_link1", end="panda_link2"))

SE3(0, 0, 0.333) ⊕ Rz(q0) ⊕ SE3(-90°, -0°, 0°) ⊕ Rz(q1)
```

which is the sequence of transforms fkine is going to compute.

From the table printed by `print(robot)` you can see that panda_link0 starts with a 0.333m transform in the z direction.

```
┌─────┬──────────────┬───────┬─────────────┬────────────────────────────────────────────────┐
│link │     link     │ joint │   parent    │              ETS: parent to link               │
├─────┼──────────────┼───────┼─────────────┼────────────────────────────────────────────────┤
│   0 │ panda_link0  │       │ BASE        │                                                │
│   1 │ panda_link1  │     0 │ panda_link0 │ SE3(0, 0, 0.333) ⊕ Rz(q0)                      │
│   2 │ panda_link2  │     1 │ panda_link1 │ SE3(-90°, -0°, 0°) ⊕ Rz(q1)                    │
│   3 │ panda_link3  │     2 │ panda_link2 │ SE3(0, -0.316, 0; 90°, -0°, 0°) ⊕ Rz(q2)       │
│   4 │ panda_link4  │     3 │ panda_link3 │ SE3(0.0825, 0, 0; 90°, -0°, 0°) ⊕ Rz(q3)       │
│   5 │ panda_link5  │     4 │ panda_link4 │ SE3(-0.0825, 0.384, 0; -90°, -0°, 0°) ⊕ Rz(q4) │
│   6 │ panda_link6  │     5 │ panda_link5 │ SE3(90°, -0°, 0°) ⊕ Rz(q5)                     │
│   7 │ panda_link7  │     6 │ panda_link6 │ SE3(0.088, 0, 0; 90°, -0°, 0°) ⊕ Rz(q6)        │
│   8 │ @panda_link8 │       │ panda_link7 │ SE3(0, 0, 0.107)                               │
└─────┴──────────────┴───────┴─────────────┴────────────────────────────────────────────────┘
```

closed 2023-04-28 by jhavl

chisarie, 2023-04-28:
Hi @jhavl ,

Thank you very much for the prompt response. Good that the system is working as you intended, than it's nothing unexpected for you.  I just found confusing that also the "parent" transform is included, but if this is the convention you choose, I can live with it.

The reason why I find this confusing is that I assumed that the transforms should be "chainable", but in this case they aren't. Roughly speaking, I expected that, 
```fkine(link0, link1) @ fkine(link1, link2) == fkine(link0, link2)```
but in this setting is not.

In fact, what we get is:
```
fkine(link0, link1)  ->  SE3(0, 0, 0.333) ⊕ Rz(q)
fkine(link1, link2)  ->  SE3(0, 0, 0.333) ⊕ Rz(q0) ⊕ SE3(-90°, -0°, 0°) ⊕ Rz(q1)
```
but  also

```
fkine(link0, link2)  ->  SE3(0, 0, 0.333) ⊕ Rz(q0) ⊕ SE3(-90°, -0°, 0°) ⊕ Rz(q1)
```

Moreover, looking at the actual visualization of the links (taken from [your blog post](https://petercorke.com/robotics/franka-emika-panda-kinematics-and-singularities/)), you can see that there should be no translation between `link1` and `link2`. 

But as I said, I guess it is just a matter of convention. I do understand that it would be a lot of effort at this point to change this convention. Still, I would suggest to at least add a heads up in the documentation, to avoid confusions.

[image]

jhavl, 2023-04-28:
Hey @chisarie, there are some things to clarify here

The parent link is not included:

panda_link0 in the URDF model is a static identity transform -- this is located at the base of the robot and corresponds to 0 in the figure above.

panda_link1 has a transform of Tz(0.333) * Rz() -- this is the transform from 0 to 1 in the figure above.

 panda_link2 has a transform of SE3(-90°, -0°, 0°) * Rz() -- this is the transform from 1 to 2 in the figure above.
 
The key to the convention is that the calculated transform will be from the **beginning** of `start` to the **end** of `end`. Therefore to get the transform of fkine(panda_link0, panda_link1) we do 

Tz(0.333) * Rz(q0) *  SE3(-90°, -0°, 0°) * Rz(q1)

Also with this line `fkine(link0, link1) @ fkine(link1, link2) == fkine(link0, link2)`, that does not work as detailed above because the transform is inclusive of the `start` and `end` links

I think the example you chose may be causing confusion as panda_link0 is a null transform (an identity) so fkine(link0, link2) == fkine(link1, link2)

it may make more sense if you do it again but with link1, link2 and link3.

chisarie, 2023-04-29:
Yup I understand, thank you very much. 

Usually I've always interpreted a link as a coordinate frame: it does not have a beginning and an end, it's just one point in space. But I now understand your interpretation and I can work with it, so thank you again for all the explanation, it has been very helpful.