petercorke/robotics-toolbox-python #369: Ctraj function does not compute the full interpolation

Hydran00, 2023-05-10:
Hi, I am trying to create trajectories in SE3 using the following code. The problem is that sometimes ctraj fails to compute the full trajectory and put some zeros. 

**To Reproduce**
```
point1 = [0.08000000548794595, 0.26799983808430966, 1.3170008551149532]
point2 = [0.08, 0.620999813079834, 1.0]
q1 = [-0.1407292785938335,0.6929612283035163, 0.6929613044483747,0.14072894781013623]
q2 = [-0.7071068286895752,-0.7071068286895752, 0, 0]

R3 = np.array(UnitQuaternion(s=q1[0],  v=q1[1:],norm=True,check=True).SE3())
R4 = np.array(UnitQuaternion(s=q2[0],  v=q2[1:],norm=True,check=True).SE3())

# Create a SE3 object from the quaternion and translation vector
T3 = SE3.Rt(R1[:3,:3], [point1[0], point1[1], point1[2]])
T4 = SE3.Rt(R2[:3,:3], [point2[0], point2[1], point2[2]])

# Interpolate between the two SE3 objects
tg = rtb.ctraj(T3, T4, 10)
for i in range(0, len(tg)):
    print(tg[i].t, " # ", tr2eul(tg[i].R))
```
with this code I get the following output
```
[0.08000001 0.26799984 1.31700086]  #  [ 1.57079756  2.74087494 -3.14159153]
[0.08000001 0.27780539 1.30819528]  #  [ 1.57079758  2.57747306 -3.14159156]
[0.08       0.30722206 1.28177854]  #  [ 1.57079766  2.08726743 -3.14159164]
[0.08       0.35624983 1.23775064]  #  [ 1.57079787  1.27025804 -3.14159185]
[0.08       0.41508316 1.18491717]  #  [1.57079936 0.28984678 3.14159198]
[0. 0. 0.]  #  [0. 0. 0.]
[0. 0. 0.]  #  [0. 0. 0.]
[0.08       0.58177759 1.03522232]  #  [-1.57079524e+00  2.48798514e+00  1.26695025e-06]
[0.08       0.61119426 1.00880558]  #  [-1.57079517e+00  2.97819078e+00  1.19740215e-06]
[0.08       0.62099981 1.        ]  #  [0.         3.14159265 1.57079633]
```
In this case two points are skipped.  
  

Another problem is that usually ctraj does not take the path fastest in orientation. This maybe can be solved checking dot product of quaternions. I found this check in the slerp implementation of PyQuaternion library: 

```
# If the dot product is negative, slerp won't take the shorter path.
# Note that v1 and -v1 are equivalent when the negation is applied to all four components.
# Fix by reversing one quaternion
dot = np.dot(q0.q, q1.q)
if dot < 0.0:
    q0.q = -q0.q
    dot = -dot
```

Hydran00, 2023-05-19:
Thank you for your response. In the code I posted I put the wrong quaternion and rotations matrices were R1 and R2 as you mentioned. The problematic quaternion was  :
```q2 = [0,-0.7071068286895752, -0.7071068286895752, 0]```
Anyway, after upgrading the package the issue seems to be solved and I get the right output.

petercorke, 2023-05-19:
There was for a while a bug with the conversion from rotation matrix to quaternion which might have been responsible.  I’m glad it’s working now, and we’ll try and get updates out a bit more often, and also the rate of change of the code base will slow as we approach maturity.  Thanks for your interest in this project.Sent from my iPad

closed 2023-05-20 by petercorke