petercorke/robotics-toolbox-python #372: module 'roboticstoolbox' has no attribute 'models'

WHuangEI, 2023-05-26:
my code is as follows:

import roboticstoolbox as rtb
panda = rtb.models.Panda()
panda.q = panda.qr
print(panda.jacobm(panda.q)

This error occurred. My question is how can I use the URDF file to accurately calculate the robot Jacobian matrix?

WHuangEI, 2023-05-27:
My roboticstoolbox-python version is 1.0.2, Is this problem caused by the too old version ?

WHuangEI, 2023-05-27:
I tried to update it using ' pip3 install --upgrade roboticstoolbox-python ’， and then I got : pip3 install --upgrade roboticstoolbox-python
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: roboticstoolbox-python in ./.local/lib/python3.6/site-packages (1.0.2)
Requirement already satisfied: swift-sim~=1.0.0 in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (1.0.0)
Requirement already satisfied: spatialmath-python~=1.0.0 in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (1.0.5)
Requirement already satisfied: spatialgeometry~=1.0.0 in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (1.0.1)
Requirement already satisfied: pgraph-python in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (0.6.2)
Requirement already satisfied: progress in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (1.6)
Requirement already satisfied: matplotlib in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (3.3.4)
Requirement already satisfied: rtb-data in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (1.0.1)
Requirement already satisfied: ansitable in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (0.9.7)
Requirement already satisfied: numpy>=1.17.4 in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (1.19.5)
Requirement already satisfied: scipy in ./.local/lib/python3.6/site-packages (from roboticstoolbox-python) (1.5.4)
Requirement already satisfied: colored in ./.local/lib/python3.6/site-packages (from spatialmath-python~=1.0.0->roboticstoolbox-python) (1.4.4)
Requirement already satisfied: websockets in ./.local/lib/python3.6/site-packages (from swift-sim~=1.0.0->roboticstoolbox-python) (9.1)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.3 in /usr/local/lib/python3.6/dist-packages (from matplotlib->roboticstoolbox-python) (3.0.9)
Requirement already satisfied: pillow>=6.2.0 in ./.local/lib/python3.6/site-packages (from matplotlib->roboticstoolbox-python) (8.4.0)
Requirement already satisfied: cycler>=0.10 in ./.local/lib/python3.6/site-packages (from matplotlib->roboticstoolbox-python) (0.11.0)
Requirement already satisfied: kiwisolver>=1.0.1 in ./.local/lib/python3.6/site-packages (from matplotlib->roboticstoolbox-python) (1.3.1)
Requirement already satisfied: python-dateutil>=2.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib->roboticstoolbox-python) (2.8.2)
Requirement already satisfied: six>=1.5 in /usr/lib/python3/dist-packages (from python-dateutil>=2.1->matplotlib->roboticstoolbox-python) (1.11.0)

This shows that I did not update it successfully.  but why ?

WHuangEI, 2023-05-27:
And how i can get the jacobian matrix from a URDF file defined by myself.  Will be grateful for any help.

jhavl, 2024-06-19:
Hi @WHuangEI, my only guess for this issue is that you are using Python 3.6 which is locking you to an old and unsupported version of the toolbox. I'd advise you to try a later version of Python 3.8 and try again.

closed 2024-06-19 by jhavl