petercorke/robotics-toolbox-python #376: Collision detection

jianchaoci, 2023-05-31:
Hi,

I want to use roboticstoolbox to do collision detection. But I got a problem when creating the obstacles. I follow the instruction you give to crate box using rtb.Box(). But I got an error as below.

[image]

jkaniuka, 2023-06-11:
Hi @jianchaoci :wave:, I've attached the working code below

```python
import roboticstoolbox as rtb
from spatialmath import *
from spatialgeometry import *

panda = rtb.models.Panda()
obstacle = Cuboid([1, 1, 1], pose = SE3(1, 0, 0))
iscollision = panda.iscollided(panda.q, obstacle) # boolean
iscollision = panda.links[0].iscollided(obstacle)
```

jianchaoci, 2023-06-12:
> Hi @jianchaoci 👋, I've attached the working code below
> 
> ```python
> import roboticstoolbox as rtb
> from spatialmath import *
> from spatialgeometry import *
> 
> panda = rtb.models.Panda()
> obstacle = Cuboid([1, 1, 1], pose = SE3(1, 0, 0))
> iscollision = panda.iscollided(panda.q, obstacle) # boolean
> iscollision = panda.links[0].iscollided(obstacle)
> ```

Hi Jkaniuka,
Really thanks for your reply. I am wondering is there a method to help to visualize the created obstacle. And is there a IK_xx method to help to avoid those solutions collide the obstacle?

jkaniuka, 2023-06-12:
@jianchaoci, glad I could help :wink: I've only used _robotics-toolbox-python_ to generate manipulator trajectories, so I don't know if it has the functions you're talking about. I can, however, offer you other solutions :smile:

- Robotics Toolbox for Python uses [**PyBullet**](https://github.com/bulletphysics/bullet3) for collision checking. Using this module, you can easily visualize the obstacles you add. Besides, you can check there whether the robot collides with an obstacle. You can find sample code in this tutorial - [**Collision Detection in PyBullet**](https://adamheins.com/blog/collision-detection-pybullet).
- If you want to generate a trajectory so that the robot does not come into collision with an obstacle, you can use [**MoveIt**](https://moveit.picknik.ai/main/index.html). MoveIt is motion planning framework that is part of [**ROS 2**](https://docs.ros.org/en/humble/index.html). If you're familiar with _Robot Operating System_, I suggest you check this tutorial - [**Planning Around Objects**](https://moveit.picknik.ai/main/doc/tutorials/planning_around_objects/planning_around_objects.html).

closed 2024-06-19 by jhavl