petercorke/robotics-toolbox-python #383: Swift crashes before plotting the robot

J2M2, 2023-06-17:
**Describe the bug**
I am experiencing an issue while trying to display a the robot plot in 3D using the roboticstoolbox library. I am following the code examples mentioned [here](https://github.com/jhavl/swift#code-examples), but I encounter errors related to file path resolution and file not found when trying to load mesh files (.dae) for visualization.

During the plot generation, I encounter the following error message related to file path resolution and file not found:

```
Error: Could not load retrieve/[file-path]: fetch for "[http://localhost:52000/retrieve/http://localhost:52000/retrieve/C:/Users/juanm/Dev/Robot/Toolbox/.venv/Lib/site-packages/rtbdata/xacro/franka_description/meshes/visual/link1.dae]" responded with 404: File not found.
```

I have check manually if the path exist, and I can access to the .dae file. However the complete path (including /localhost:52000/retrieve/... gives me 404)

**Version information**

Installed using PyPI inside of a Python (3.10.8) .venv:

- roboticstoolbox-python (1.1.0)
- swift-sim (1.1.0)

**To Reproduce**
Steps to reproduce the behavior:

1. Use a local Python environment and install roboticstoolbox-python (in Windows)
2. Run the following script:
```python
import roboticstoolbox as rp

panda = rp.models.Panda()
panda.plot(q=panda.qr)
```
3. Refresh the page and get an empty 3D view for few seconds and it crashes. The console output gives:

```
connection handler failed
Traceback (most recent call last):
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\websockets\legacy\protocol.py", line 959, in transfer_data
    message = await self.read_message()
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\websockets\legacy\protocol.py", line 1029, in read_message
    frame = await self.read_data_frame(max_size=self.max_size)
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\websockets\legacy\protocol.py", line 1104, in read_data_frame
    frame = await self.read_frame(max_size)
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\websockets\legacy\protocol.py", line 1161, in read_frame
    frame = await Frame.read(
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\websockets\legacy\framing.py", line 95, in read
    mask_bits = await reader(4)
  File "C:\Users\juanm\AppData\Local\Programs\Python\Python310\lib\asyncio\streams.py", line 696, in readexactly
    raise self._exception
  File "C:\Users\juanm\AppData\Local\Programs\Python\Python310\lib\asyncio\proactor_events.py", line 301, in _loop_reading
    self._read_fut = self._loop._proactor.recv_into(self._sock, self._data)
  File "C:\Users\juanm\AppData\Local\Programs\Python\Python310\lib\asyncio\windows_events.py", line 477, in recv_into
    ov.WSARecvInto(conn.fileno(), buf, flags)
ConnectionAbortedError: [WinError 10053] Se ha anulado una conexión establecida por el software en su equipo host

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\websockets\legacy\server.py", line 240, in handler
    await self.ws_handler(self)
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\websockets\legacy\server.py", line 1186, in _ws_handler
    return await cast(
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\swift\SwiftRoute.py", line 320, in serve
    await self.expect_message(websocket, expected)
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\swift\SwiftRoute.py", line 325, in expect_message
    recieved = await websocket.recv()
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\websockets\legacy\protocol.py", line 568, in recv
    await self.ensure_open()
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\websockets\legacy\protocol.py", line 935, in ensure_open
    raise self.connection_closed_exc()
websockets.exceptions.ConnectionClosedError: no close frame received or sent
Traceback (most recent call last):
  File "c:\Users\juanm\Dev\Robot\Toolbox\swift_test.py", line 4, in <module>
    panda.plot(q=panda.qr)
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\roboticstoolbox\robot\BaseRobot.py", line 2342, in plot
    env.add(self, readonly=True, **kwargs)
  File "C:\Users\juanm\Dev\Robot\Toolbox\.venv\lib\site-packages\swift\Swift.py", line 405, in add
    while not int(self._send_socket("shape_mounted", [id, len(robob)])):
ValueError: invalid literal for int() with base 10: 'Connected'
```

**Expected behavior**
I expected the roboticstoolbox library to automatically handle the loading and visualization of the robot model using the specified URDF file and associated mesh files (STL or Collada format). The tutorial implies that the visualization should work seamlessly.

**Screenshots**
[image]

**Environment (please complete the following information):**
 - Windows 11
- Python 3.10.8

johannkipping, 2023-07-04:
I have the exact same issue, in the swift repository there is another thread that also mentions this. Any help would be appreciated!

GEYOUR, 2023-07-08:
Same issue here.

jacobvartanian, 2023-08-03:
Was having a similar issue, where it would work on Ubuntu but not on Windows. A "quick fix" for Windows, from the swift library, in the file `SwiftRoute.py`, change the following block to have `self.path[10:]` instead of `self.path[9:]`

Currently, `self.path` is being modified to retain the `/` character at the start of the path (which is what you would expect on Linux based systems. On Windows, this is resulting in the path being `/C:/Users/....` which seems to be causing the issue.

```
elif self.path.startswith("/retrieve/"):
    # print(f"Retrieving file: {self.path[10:]}")
    self.path = urllib.parse.unquote(self.path[9:])
    self.send_file_via_real_path()
    return
```

See this comment here: https://github.com/jhavl/swift/commit/f6c8cdea481b713540b5889c0c8c125ca0cc3ac0#r116060265

J2M2, 2023-08-03:
[image]

@jacobvartanian This `quickfix` worked for me. (Of course I had to to the change directly in the library of the .venv but is ok as temporaly solution.

@jhavl Will you do a patch for this?

KuabeM, 2023-10-20:
I encountered the same error. I opened a PR with this fix: jhavl/swift#52, let's see if this gets merged

petercorke, 2026-08-25:
Sorry for the long wait on this one — I know it's been open since 2023 with several of you piling on with workarounds.

I looked into this in depth against swift's current codebase (see the investigation on jhavl/swift#52, which is the specific patch several of you had been applying manually). Short version: swift's frontend and backend were both rewritten from scratch since this issue was filed, and the URL construction that builds the `/retrieve/...` path changed along with it — the exact line this issue's fix targeted is now correct for how URLs are built today, so applying the old patch as-is would actually break the currently-working Linux/Mac case rather than fix anything.

That said, the underlying Windows concern may still be real in a different shape: drive-letter paths (`C:/Users/...`) aren't normalized before being sent to the frontend, so they may not survive the `/retrieve/` round-trip cleanly even now. If you're still hitting a failure like this on the current swift-sim release (2.0.0+) and roboticstoolbox 1.4.0+, please open a fresh issue with a repro against the current code — that'll make it much easier to pin down whatever's left, since the code this issue was originally filed against no longer exists.

closed 2026-08-25 by petercorke