petercorke/robotics-toolbox-python #384: Broken Install

kingjin94, 2023-06-19:
**Describe the bug**
One of the dependencies (colored) seems to have changed and breaks the import of parts of the toolbox (e.g. `from roboticstoolbox import mstraj` on pythono >= 3.9)

**Version information**

Installed with PyPI

```
$ pip list

Package                Version
---------------------- -------
ansitable              0.9.7
colored                1.5.0
contourpy              1.1.0
cycler                 0.11.0
fonttools              4.40.0
kiwisolver             1.4.4
matplotlib             3.7.1
numpy                  1.25.0
packaging              23.1
pgraph-python          0.6.2
Pillow                 9.5.0
pip                    22.3.1
progress               1.6
pyparsing              3.1.0
python-dateutil        2.8.2
roboticstoolbox-python 1.1.0
rtb-data               1.0.1
scipy                  1.10.1
setuptools             65.5.0
six                    1.16.0
spatialgeometry        1.1.0
spatialmath-python     1.1.8
swift-sim              1.1.0
typing_extensions      4.6.3
websockets             11.0.3
wheel                  0.38.4
```

**To Reproduce**
```
docker run -ti python:3.10 bash
pip install roboticstoolbox-python
python
from roboticstoolbox import mstraj
```
Results in:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/site-packages/roboticstoolbox/__init__.py", line 5, in <module>
    from roboticstoolbox.mobile import *
  File "/usr/local/lib/python3.10/site-packages/roboticstoolbox/mobile/__init__.py", line 6, in <module>
    from roboticstoolbox.mobile.PlannerBase import PlannerBase
  File "/usr/local/lib/python3.10/site-packages/roboticstoolbox/mobile/PlannerBase.py", line 24, in <module>
    from colored import fg, attr
ImportError: cannot import name 'fg' from 'colored' (/usr/local/lib/python3.10/site-packages/colored/__init__.py)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
Linux (standard python docker image) with:

```
$ cat /etc/os-release 

PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

kingjin94, 2023-06-19:
Hotfix could be to downgrade dependency: `pip install -U "colored<1.5"`

sdiebolt, 2023-06-19:
This is a problem of colored: they introduced breaking API changes in 1.5.0 and renamed functions `fg` to `fore` and `attr` to `style`.

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking just over three years to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one: this was caused by the `colored` dependency renaming its API (`fg`/`attr` → `fore`/`style`) in a breaking release. RTB's `pyproject.toml` now pins `colored>2.0.0`, matching the renamed API, so a normal `pip install roboticstoolbox-python` should resolve a compatible version automatically.

While I can't promise every corner of this has been exercised, I'd like to ask you to retry it (if you're still motivated) and get back to me with a comment here, or a new issue, if the problem remains.

closed 2026-08-25 by petercorke