petercorke/robotics-toolbox-python #393: Inconsistent joint indexes in ETS

agudym, 2023-07-22:
**Describe the bug**
Multiplication (concatenation) of several ETS does not update joint indexes

**Version information**

roboticstoolbox-python 1.1.0

**To Reproduce**
```python
import roboticstoolbox as rtb

et = rtb.ET.Rz()
ets = rtb.ETS([et, ])

print((et * et).joints()) # OK:
# [ET.Rz(jindex=0), ET.Rz(jindex=1)]

print((ets * ets).joints()) # NOT OK:
# [ET.Rz(jindex=0), ET.Rz(jindex=0)]

print((et * et *  ets).joints()) # NOT OK:
# [ET.Rz(jindex=0), ET.Rz(jindex=1), ET.Rz(jindex=0)]

print((et *  ets).joints()) # NOT OK:
# raise ValueError( "You can not have some jindices set for the ET's in arg. It must be all")
```
**Expected behavior**
ETS consctructor show re-enumerate joint indexes

**Environment (please complete the following information):**
 - Ubuntu 22.04
 - python 3.10.6

p.s. Awesome work btw :) !!!

agudym, 2023-07-22:
If that's a bug (mb it's some "feature"?), I would be glad to help with the investigation

smoltafet, 2026-10-25:
I would like to work on this. The cause is in `ETS.__init__`: when ETS objects are concatenated, the ETs arrive already carrying the jindices that were automatically assigned inside their source ETS, so the combined sequence either keeps duplicate indices (all set, kept verbatim) or raises the all-or-none error when a bare ET is mixed in. My plan is to track whether the combined sequence came entirely from automatically indexed ETSs (the existing `_auto_jindex` flag) plus unindexed ETs, and only then drop the inherited jindices so they are re-derived for the combined chain. Explicitly assigned jindices would be preserved exactly as now. The same code path exists in ETS2, so I would fix it there too. Tests for the four combinations from the report plus an explicit-jindex preservation case.

smoltafet, 2026-10-25:
Update after investigating: I am not going to file the renumbering change I described above, because it breaks behaviour the test suite locks in. `tests/test_ETS.py::test_jacob0` and `test_jacobe` build six single-joint ETSs (`ETS(tx())`, `ETS(Rx())`, ...) and combine them with `+`; the combined ETS is then used with a length-1 `q`, which only works because every joint keeps jindex 0, i.e. the joints share one coordinate and `jacob0`/`jacobe` assemble one column per joint using `q[jindex]` as its value. The `_sym` variants use the same pattern with explicit jindices. Renumbering auto-assigned jindices on concatenation makes those tests fail with `q has length 1, ...`.

So the current keep-the-jindices behaviour is load-bearing for the Jacobian-assembly use case, and it conflicts with the serial-chain expectation in this issue (where `ets * ets` producing two joints that both read `q[0]` is surprising). Which composition should win, or should the two be distinguished somehow (for example `*` vs `+`, which are currently aliased)? Happy to implement whichever direction you prefer; I did not want to guess at the semantics.