petercorke/robotics-toolbox-python #409: Can't use the swift simulator

WHuangEI, 2023-12-26:
**Check here first**

[Common issues](https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues)

**Describe the bug**
when I just run the 'holistic_mm_non_holonomic.py', the browser shows nothing but :
The connection was reset
    The site could be temporarily unavailable or too busy. Try again in a few moments.
    If you are unable to load any pages, check your computer’s network connection.
    If your computer or network is protected by a firewall or proxy, make sure that Firefox is permitted to access the web.
and the information in terminal:
$ python holistic_mm_non_holonomic.py 
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 45358)
Traceback (most recent call last):
  File "/usr/lib/python3.6/socketserver.py", line 320, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 351, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 364, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/hwww/.local/lib/python3.6/site-packages/swift/SwiftRoute.py", line 222, in __init__
    *args, directory=str(root_dir), **kwargs
TypeError: __init__() got an unexpected keyword argument 'directory'
----------------------------------------
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 45372)
Traceback (most recent call last):
  File "/usr/lib/python3.6/socketserver.py", line 320, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 351, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 364, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/hwww/.local/lib/python3.6/site-packages/swift/SwiftRoute.py", line 222, in __init__
    *args, directory=str(root_dir), **kwargs
TypeError: __init__() got an unexpected keyword argument 'directory'
----------------------------------------
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 45380)
Traceback (most recent call last):
  File "/usr/lib/python3.6/socketserver.py", line 320, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 351, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 364, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/hwww/.local/lib/python3.6/site-packages/swift/SwiftRoute.py", line 222, in __init__
    *args, directory=str(root_dir), **kwargs
TypeError: __init__() got an unexpected keyword argument 'directory'
----------------------------------------
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 45390)
Traceback (most recent call last):
  File "/usr/lib/python3.6/socketserver.py", line 320, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 351, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 364, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/hwww/.local/lib/python3.6/site-packages/swift/SwiftRoute.py", line 222, in __init__
    *args, directory=str(root_dir), **kwargs
TypeError: __init__() got an unexpected keyword argument 'directory'
----------------------------------------
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 45394)
[... 105 lines cut ...]

Did you install from PyPI or GitHub?
If PyPI what version number?  
PyPI  pip 21.3.1
If GitHub what commit hash?

Robotics Toolbox depends heavily on two other packages: Swift (3D graphics) and SpatialMath toolbox (underpinning maths utilities).  If you think your issue is related to these, then please answer the questions above for them.

**To Reproduce**
Steps to reproduce the behavior:
1. The shortest, complete, Python script that exhibits the bug.
2. The script output, including error messages.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Your OS (MacOS, Linux, Windows). ubuntu 18.04
- Your Python version.  python 3.6

**Additional context**
Add any other context about the problem here.

WHuangEI, 2023-12-26:
How can I solve this, it is important for me.

WHuangEI, 2023-12-26:
the version i checked:
swift -sim:1.0.0
spatialgeometry:1.0.1
spatialmath-python:1.0.5
PyPI pip 21.3.1
ubuntu 18.04, python 3.6

WHuangEI, 2023-12-26:
@jhavl

petercorke, 2026-08-25:
Sorry for the long wait getting back to you on this one.

This is a Python 3.6 incompatibility — `socketserver.SimpleHTTPRequestHandler`'s `directory=` keyword argument wasn't added until Python 3.7, and swift's server code has needed it since. Python 3.6 reached end-of-life in December 2021, and current roboticstoolbox/swift-sim both require a much newer Python (3.10+) — so this specific failure mode can't occur on any supported environment today.

Closing as an unsupported/obsolete environment issue. If you hit something similar on a current Python version, please open a fresh issue with the details.

closed 2026-08-25 by petercorke