petercorke/robotics-toolbox-python #412: Import Error: randn from scripy

NicholasSnyder, 2024-01-21:
**Describe the bug**
ImportError: cannot import name 'randn' from 'scipy'

**Version information**
installed from pip 23.2.1

**To Reproduce**
Steps to reproduce the behavior:

```
from spatialmath.base import *
from roboticstoolbox import *
```

OUTPUT:
Exception has occurred: ImportError
cannot import name 'randn' from 'scipy' (C:\Users\Nicholas Snyder\AppData\Local\Programs\Python\Python312\Lib\site-packages\scipy\__init__.py)
  File "C:\Users\Nicholas Snyder\Downloads\RVC3-python-main\test.py", line 2, in <module>
    from roboticstoolbox import *
ImportError: cannot import name 'randn' from 'scipy' (C:\Users\Nicholas Snyder\AppData\Local\Programs\Python\Python312\Lib\site-packages\scipy\__init__.py)

**Expected behavior**
Expected toolbox to successfully import. 

**Screenshots**
[image]

**Environment**
 - Windows Version	10.0.19045 Build 19045
- Python 3.12.1 64bit

**Additional context**
The only thing I was able to find on this issue is that scipy does not have a random module to import anymore. A proposed alternative is to use random from numpy. However, I am not sure if this is the actual underlying issue.

kingjin94, 2024-01-22:
Can confirm this error on Python 3.9/10/11 in the official Python containers (https://hub.docker.com/_/python). "pip install scipy<1.12.0" seems to be a viable hot fix.

FelipeCybis, 2024-01-22:
It turns out the `scipy.randn` function was not even used by `robotics-toolbox-python`. I created a PR to fix this.

WendaXu2023, 2024-02-12:
I meet the same problem. Is it fixed?

wanglong06, 2024-02-13:
I am currently having the same problem, and currently, I confirm that @kingjin94 's proposed hot fix (scipy<1.12.0) work. But a proper fix would be greatly appreciated!

JakobDomislovic, 2024-03-19:
Go to roboticstoolbox/mobile/EKF.py and change that randn is imported from numpy and remove it from scipy import. Use: from numpy.random import randn

daironxuan, 2024-03-24:
### site-packages/roboticstoolbox/mobile/EKF.py:

```python
from collections import namedtuple
import numpy as np
from math import pi
from scipy import integrate
from scipy.linalg import sqrtm, block_diag
from scipy.stats.distributions import chi2
import matplotlib.pyplot as plt
from matplotlib import animation

from spatialmath.base.animate import Animate
from spatialmath import base, SE2
from roboticstoolbox.mobile import VehicleBase
from roboticstoolbox.mobile.landmarkmap import LandmarkMap
from roboticstoolbox.mobile.sensors import SensorBase

# Use numpy's randn instead of scipy's
from numpy.random import randn
```

pappuyadav, 2024-05-07:
> I meet the same problem. Is it fixed?

https://github.com/petercorke/robotics-toolbox-python/issues/436

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking close to two and a half years to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one: this was caused by an old `from scipy import randn`-style import that SciPy removed. I checked just now — RTB's own code no longer touches that import (it's commented out, dead), so a fresh install shouldn't hit this anymore.

While I can't promise every corner of this has been exercised, I'd like to ask you to retry it (if you're still motivated) and get back to me with a comment here, or a new issue, if the problem remains.

closed 2026-08-25 by petercorke