petercorke/robotics-toolbox-python #417: Is there a function to load a URDF to a DHRobot? Or alternatively a function to back out the DH parameters from an ERobot loaded from a URDF?

jpenaloza1211, 2024-02-14:
**Describe the bug**
I am working on loading custom robots using URDFs and would like to be able to create both versions from a single URDF for my work.

**Version information**
PyPI roboticstoolbox-python 1.1.0

**To Reproduce**
I've tried to load in a URDF as an ERobot and then convert to DHRobot by passing in the links but the DHRobot class expects DHLinks. 

`robot_path = os.path.join(os.getcwd(), 'models')
    UR5 = UR5eCustom(robot_path)
    UR5_dh = DHRobot(UR5.links)`

I've also tried to create my robot class based on the DHRobot class but the DHRobot class doesn't take in the URDF and convert it to DHLinks.

**Expected behavior**
Just looking for a function to be able to easily switch between a DHRobot and an ERobot or a way to create both types of robots from the same URDF.

**Environment (please complete the following information):**
 - Ubuntu 22.04
- Python 3.10

**Additional comment**
I've seen papers for switching between ets and dh, but want to ask if that functionality exists in this library before resorting to that.

orat, 2025-02-26:
I am also interesting in that:-)

petercorke, 2026-08-25:
Sorry for the slow reply, and thanks for the detailed question.

Short answer: no, there isn't (and still isn't) a built-in function to automatically convert between a URDF/ERobot's kinematic representation (ETS) and DH parameters. Deriving DH parameters from an arbitrary ETS chain isn't a simple mechanical conversion — DH parameters only exist for chains that satisfy specific structural constraints, and a general URDF-derived ETS chain doesn't always fit that.

There is some prior art, though: the MATLAB version of Robotics Toolbox has a function called `DHFactor`, written in Java, that attempts to factorize an ETS string into DH parameters. Not all ETS strings can be factorized this way, and even where a valid factorization exists, `DHFactor` isn't guaranteed to find it. A Python port of this is on the back burner but not yet in a state to release.

In the meantime, the practical approach is still to build the `DHRobot` and `ERobot` versions of a given manipulator separately from their own parameter/URDF sources — as most of the bundled models do (see e.g. `models.DH.Panda` and `models.URDF.Panda`).

Closing since this reflects the current capability of the toolbox — happy to reopen if you'd like to discuss further or this is blocking something specific.

closed 2026-08-25 by petercorke