petercorke/robotics-toolbox-python #434: Failed to run examples/neo.py

EstanislauFilho, 2024-05-02:
The error states that the dimensions of the matrices you are trying to concatenate are different along axis 1. The array at position 0 has size 13 and the array at position 1 has size 15. To use the np.r_ function, the dimensions along axes except the concatenation axis must be identical.

I attempted to brute force resize the matrices to ensure that the dimensions of the matrices were compatible. However, I was unsuccessful.

[image]

haha1227, 2024-06-17:
The same problem. Did you solve this problem? thx

ybl2333, 2024-06-25:
I have same question and try to install 1.0.3, it's work.

pogosjan12, 2024-07-24:
@ybl2333 What 1.0.3 ? Is that the version of the rtb or something else that you changed ?

adriencholet, 2025-02-28:
I had the same issue, it comes from a mismatch in the dimensions of the collision constraints array and the joint velocity constraints array. I fixed it by modifying line 113 of neo.py:
`c_Ain = np.c_[c_Ain, np.zeros((c_Ain.shape[0], 6))]`
to
`c_Ain = np.c_[c_Ain, np.zeros((c_Ain.shape[0], 4))]`

petercorke, 2026-09-19:
Thank you for your interest in RTB, and sorry for the trouble getting the example running! This is fixed — `joint_velocity_damper` had a keyword-argument bug in `examples/neo.py` that's now corrected (#679, #690). Thanks for the report.

closed 2026-09-19 by petercorke