petercorke/robotics-toolbox-python #439: Is it possible to detect when a Gripper collides with an obstacle?

EstanislauFilho, 2024-05-12:
I would like to know if it is possible to detect when a Gripper has collided with a certain obstacle. I am using a Kinova Gen3 manipulator and need to know if the gripper has collided with a specific obstacle. In the image below, I demonstrate a situation where the robotiq_2f_85 gripper is in contact with the obstacle, but the algorithm cannot detect the obstacle.

I noticed in the source code that the Gripper.py class does not have any collision detection method. Is there an easy way to perform this detection, or do I need to implement this code myself?

[image]

TJMolnar, 2024-05-21:
I am also having this issue with the Kinova and I have narrowed it down to a limitation in the [spatialgeometry](https://github.com/jhavl/spatialgeometry) package used by this toolbox, which itself relies on pybullet to handle collision meshes from the URDF. Kinova provides Collada (.dae) mesh files for their arm, but pybullet does not seem to support them in this context. Indeed the spatialgeometry library is explicitly only loading .stl files, and so is actually ignoring these meshes entirely. Collada files are a very common format for meshes in URDF's and are (as in this case) often provided in official packages from manufacturers, so this places a significant limitation on the usage of this toolbox.

EstanislauFilho, 2024-08-04:
@TJMolnar I am trying to check for situations where the Kinova Gen3 manipulator collided with itself. Do you know any simple way to verify this situation? I tried a few things but nothing worked.