petercorke/robotics-toolbox-python #444: How to install vpython as backend?

NirViaje, 2024-06-02:
**Check here first**

[Common issues](https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues)

**Describe the bug**

Traceback (most recent call last):
  File "rtb_neo.py", line 51, in <module>
    robot.plot(qt.q, backend=args.backend)
  File "/home/nirj/miniconda3/lib/python3.8/site-packages/roboticstoolbox/robot/BaseRobot.py", line 2331, in plot
    env = self._get_graphical_backend(backend)
  File "/home/nirj/miniconda3/lib/python3.8/site-packages/roboticstoolbox/robot/BaseRobot.py", line 2230, in _get_graphical_backend
    raise ValueError("unknown backend", backend)  # pragma nocover
ValueError: ('unknown backend', 'vpython')

**Version information**

rtb 1.1.0
python 3.8.10

Did you install from PyPI or GitHub?
If PyPI what version number?
If GitHub what commit hash?

Robotics Toolbox depends heavily on two other packages: Swift (3D graphics) and SpatialMath toolbox (underpinning maths utilities).  If you think your issue is related to these, then please answer the questions above for them.

**To Reproduce**
Steps to reproduce the behavior:
1. The shortest, complete, Python script that exhibits the bug.
2. The script output, including error messages.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Your OS (MacOS, Linux, Windows).
- Your Python version.

**Additional context**
Add any other context about the problem here.

a-camarillo, 2024-06-21:
It looks like vpython is not a supported backend.

https://github.com/petercorke/robotics-toolbox-python/blob/7f1bcae7e17360ea9b30a49731c2c93a219c6a19/roboticstoolbox/robot/BaseRobot.py#L2175-L2178

If vpython is globally installed through pip my suggestion is to try installing `rtb` in a virtual environment and see if that separation from your global dependencies fixes the issue

petercorke, 2026-09-19:
Thank you for your interest in RTB! vpython is no longer a supported backend — we'd recommend using Swift instead for 3D visualization. Closing since this was already flagged in the thread, but let us know if you're still stuck.

closed 2026-09-19 by petercorke