petercorke/robotics-toolbox-python #446: Pypi install V1.1.1

peter-manzl, 2024-06-13:
Hi everyone! 

I just noticed yesterday that for the newest version (V 1.1.1 from May 11th, 2024) on pypi, only the source code or the MacOS version is available. When I run (windows / Python 3.11 anaconda environment): 
pip install roboticstoolbox-python
I get a very long error message which ends with:
[image]
As far as I can see this is because the package cannot be compiled without the Microsoft Visual C++ compiler. I tried it on another machine with the compiler available and everything works there fine.

Would it be possible to either upload the other wheels or mark Version 1.1.1 as pre-release? When running 
pip install roboticstoolbox-python==1.1.0
everything works fine for me. 

Best regards, 
Peter

laorhy, 2025-01-14:
I encountered the same issue while trying to pip install roboticstoolbox-python on Windows. Thanks to pip install roboticstoolbox-python==1.1.0, it worked.

gbattocletti, 2025-01-17:
As far as I know the issue is related to the version requirements of spatialmath-python, and it _should_ have been fixed in version 1.1.13 of spatialmath-python. However, today I tried an installation of spatialmath-python on windows + python 3.13 which resulted in the same error reported by @peter-manzl (see https://github.com/bdaiinstitute/spatialmath-python/pull/150#issuecomment-2598675017).

As a workaround there are the fix suggested by @laorhy, and also two other options, namely, 1) to use a lower version of python (up to and including 3.10 the installation works fine), and 2) on any python version, install the package with the `--no-deps` specification and then add the required packages manually.

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking just over two years to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one: at the time, that release didn't have a Windows wheel published, so pip fell back to building from source and hit a spatialmath-python version mismatch. RTB now ships prebuilt wheels (via cibuildwheel) for Python 3.10 through 3.14 on Linux, macOS, and Windows for every release, so this specific gap shouldn't recur.

While I can't promise every corner of this has been exercised, I'd like to ask you to retry it (if you're still motivated) and get back to me with a comment here, or a new issue, if the problem remains.

closed 2026-08-25 by petercorke

peter-manzl, 2026-08-27:
Thank you very much for coming back to this issue and updating the toolbox.
I tested reinstalling the Python package of the toolbox in both Python 3.11 and 3.13. I did not encounter the issue again and all the scripts I was previously running (forward/inverse kinematics and visualization) work.