petercorke/robotics-toolbox-python #449: import error, cannot import name plotvol3

slyandsmart, 2024-07-04:
**Check here first**

[Common issues](https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues)

**Describe the bug**

I try to use the toolbox, installed it via pip on my DOCKER Container but the following error appears when i try to work with.

>>> import roboticstoolbox
/usr/local/lib/python3.10/dist-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
  warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/roboticstoolbox/__init__.py", line 1, in <module>
    from roboticstoolbox.tools import *
  File "/usr/local/lib/python3.10/dist-packages/roboticstoolbox/tools/__init__.py", line 2, in <module>
    from roboticstoolbox.tools.p_servo import p_servo, angle_axis, angle_axis_python
  File "/usr/local/lib/python3.10/dist-packages/roboticstoolbox/tools/p_servo.py", line 4, in <module>
    from spatialmath import SE3, base
  File "/usr/local/lib/python3.10/dist-packages/spatialmath/__init__.py", line 3, in <module>
    from spatialmath.pose2d import SO2, SE2
  File "/usr/local/lib/python3.10/dist-packages/spatialmath/pose2d.py", line 26, in <module>
    import spatialmath.base as smb
  File "/usr/local/lib/python3.10/dist-packages/spatialmath/base/__init__.py", line 8, in <module>
    from spatialmath.base.transforms3d import *  # lgtm [py/polluting-import]
  File "/usr/local/lib/python3.10/dist-packages/spatialmath/base/transforms3d.py", line 48, in <module>
    from spatialmath.base.graphics import plotvol3, axes_logic
ImportError: cannot import name 'plotvol3' from 'spatialmath.base.graphics' (/usr/local/lib/python3.10/dist-packages/spatialmath/base/graphics.py)

**Version information**

Did you install from PyPI or GitHub?
If PyPI what version number?
roboticstoolbox-python==1.1.0 (but tried all the newer ones also) 
spatialmath-python==1.1.8 (but tried all the newer ones also )

**Environment (please complete the following information):**
Ubuntu 22.04 Docker Container ( FROM osrf/ros:humble-desktop)
Python 3.10.12

peteflorence, 2024-07-14:
If not using graphics, a hacky workaround is to just not import that line, i.e. in OP's example:

`vi /usr/local/lib/python3.10/dist-packages/spatialmath/base/transforms3d.py`

and comment out line 48

petercorke, 2026-09-19:
Thank you for your interest in RTB! This traced back to a stale `spatialmath-python` version resolving under a mismatched Docker layer, not an RTB bug — RTB's minimum `spatialmath-python` requirement has since been raised, and `plotvol3` imports cleanly with current dependency versions. Closing as not reproducible on a fresh install; let us know if you still hit this.

closed 2026-09-19 by petercorke