petercorke/robotics-toolbox-python #452: Inverse kinematics with a custom tool frame

guri-dominic, 2024-07-31:
Thank you for your work on this great software package!

I am working with the xarm6 urdf with no end-effector, and I would like to use a tool frame transform. For forward kinematics, I am using
```python
Tool = sm.SE3.Tz(0.1)
ee = xarm.fkine(xarm.q, tool=Tool)
``` 
How do I include `Tool` in the ik solvers? Setting `xarm.tool = Tool` does not work, and there is no `tool` argument in the ik functions.

guri-dominic, 2024-08-01:
I found a solution:
```python
import roboticstoolbox as rbt
import spatialmath as sm

ets_tool_transform = rbt.ET.tz(0.1)

x6 = # (type = rtb.Robot), from URDF file
ets = x6.ets()
# ets = x6.ets().copy()   # could be better?
ets.insert(ets_tool_transform)

# Usage:
ets.fkine(q)
ets.ik_GN(T)  # T \in sm.SE3
```
I don't know if this is a bug, but using `x6.fkine(..)`, `x6.ik_GN(..)`, `x6.ik_GN(..)`, etc. does not produce the same results as calling `ets` functions, even though updating `ets` also updates `x6.ets()`.

shaashou, 2025-11-07:
Do you have a way to use the UR5 robotic arm and Robotiq grippers?

petercorke, 2026-09-20:
Thank you for your interest in RTB, and thanks for the follow-up workaround you posted too — that helped confirm exactly what was going on!

This is fixed in #697:
- `robot.tool`, once set, is now picked up automatically by `fkine`/`jacob0`/`jacobe`/`hessian0`/`hessiane` — no need to pass `tool=` explicitly every call (though you still can, and it overrides `self.tool` when given).
- `ik_LM`, `ik_NR`, `ik_GN`, `ikine_LM`, `ikine_NR`, `ikine_GN`, and `ikine_QP` now all accept a `tool=` parameter directly, so you no longer need to manually insert the tool transform into a copy of the ETS.

For your original case: `xarm.tool = Tool` now takes effect automatically, and `xarm.ik_GN(T, tool=Tool)` works directly.

closed 2026-09-20 by petercorke