petercorke/robotics-toolbox-python #454: Numpy 2.x support

trzy, 2024-09-04:
**Check here first**

[Common issues](https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues)

**Describe the bug**
I'd like to add roboticstoolbox to a project already using NumPy 2.x. However, it seems rbt depends on NumPy 1.x. Is there a way to upgrade this dependency?

**Version information**

Did you install from PyPI or GitHub? Github
If PyPI what version number?
If GitHub what commit hash? [2a8111b](https://github.com/petercorke/robotics-toolbox-python/commit/2a8111b10dd09c5ce8b8a94495c84ff27bcb6862)

Robotics Toolbox depends heavily on two other packages: Swift (3D graphics) and SpatialMath toolbox (underpinning maths utilities).  If you think your issue is related to these, then please answer the questions above for them.

**To Reproduce**
Steps to reproduce the behavior:
1. The shortest, complete, Python script that exhibits the bug.
2. The script output, including error messages.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Your OS (MacOS, Linux, Windows).
- Your Python version.

**Additional context**
Add any other context about the problem here.

trzy, 2024-09-04:
I tried creating a fresh environment and installing roboticstoolbox-python with pip still fails to work. That is, it installs just fine, but won't run because of this NumPy conflict (I'm assuming one of its dependencies is pulling in 2.x).

arsym-dev, 2024-09-23:
It seems this problem existed before in #324. It seems that roboticstoolbox has no max numpy version. I did the following to fix it:

``` bash
python -m pip uninstall numpy
python -m pip install numpy==1.26.4
```

trzy, 2024-11-30:
Thanks. What exactly is causing roboticstoolbox to rely on this ancient numpy version (as well as an ancient matplotlib)?

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking close to two years to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one: RTB now properly supports NumPy 2 — the old NumPy 1.x pin is long gone, and the whole C-extension build has been migrated to a toolchain (nanobind/scikit-build-core) that's tested against NumPy 2 in CI. `pip install roboticstoolbox-python` today should install cleanly alongside NumPy 2.

While I can't promise every corner of this has been exercised, I'd like to ask you to retry it (if you're still motivated) and get back to me with a comment here, or a new issue, if the problem remains.

closed 2026-08-25 by petercorke