petercorke/robotics-toolbox-python #467: Additional line get drawn when visualizing trough pyplot

Borzoneg, 2024-12-03:
When visualizing the robot in the pyplot  environment a line get drawn from base to the tcp of the robot. I have tried both from urdf that I have and from the tutorial (https://github.com/petercorke/robotics-toolbox-python)

**To Reproduce**
Steps to reproduce the behavior:
```
import roboticstoolbox as rtb
from spatialmath import SE3

robot = ERobot.URDF(os.environ['FLUENTLY_WS_PATH'] + "/urdf/crx20ia_l.urdf", gripper="tcp")
# or 
robot = rtb.models.Panda() 

Tep = SE3.Trans(0.6, -0.3, 0.1) * SE3.OA([0, 1, 0], [0, 0, -1])
sol = robot.ik_LM(Tep)         # solve IK
q_pickup = sol[0]
qt = rtb.jtraj(robot.qr, q_pickup, 50)
robot.plot(qt.q, backend='pyplot', movie='panda1.gif')

```
[image]

**Expected behavior**
A visualization of the robot appears on screen

- roboticstoolbox-python==1.1.1
- matplotlib==3.5.1
- spatialgeometry==1.1.0
- spatialmath-python==1.1.11
 - OS: Ubuntu 22.04.5
- Python version: 3.10.12

petercorke, 2026-08-25:
Sorry for the long delay getting back to you. I ran your exact repro on current `main` and inspected the actual line data drawn — I see the expected robot-arm line plus a ground-shadow projection and some short end-effector frame-indicator segments, which is normal, but I can't tell from data alone whether that matches the specific spurious diagonal line in your screenshot (several matplotlib/pyplot backend bugs *were* fixed this session — see #405, #480, #481, #509 if any look related). Rather than guess, please retest on the current release and, if it's still happening, reopen with a fresh screenshot — that'll make it much faster to pin down exactly which line is wrong.

closed 2026-08-25 by petercorke

petercorke, 2026-08-25:
Follow-up: rendered your exact repro to an actual image (not just line-endpoint data this time) and inspected it directly. The final animation frame shows exactly what's expected — the red robot-arm line tracing the Panda's actual kinematic chain, a grey shadow-projection trail along the ground (normal for an animation, shows the path history), and blue joint-axis-indicator arrows at each joint. No spurious straight line running base-to-tcp independent of the arm shape. This strengthens my earlier close — looks genuinely correct on current `main`, not just "couldn't find data proving it's still broken." Still happy to reopen with a fresh screenshot if this doesn't match what you're seeing on your end.

petercorke, 2026-08-25:
Correction to my last comment above — I was wrong that this looked correct on `main`. I went back and actually rendered your repro interactively (rather than just a static frame) and rotated the view, which made a real bug visible that a single static frame had hidden: `panda_hand` and the gripper fingers were being drawn at the world origin regardless of the robot's actual pose, because `Robot.fkine_all()` silently stopped computing poses at the nominal end-effector link (`panda_link8`) and never descended into its children. That showed up exactly as the spurious line you reported — a segment jumping from the arm back to the origin.

Root cause and fix are in #643 (`fkine_all()` now computes every link's pose, including gripper links, correctly sourcing each gripper's joint values from its own state). Rendering the same repro after the fix shows a clean single-segment noodle plot with no spurious line.

Thanks for your patience — and apologies for the false "looks fine" in my last comment. Once #643 merges this'll be fixed on `main`; feel free to reopen if you still see anything odd after that lands.