petercorke/robotics-toolbox-python #469: Memory Leak with multiple DH_robot initializations

shubh-arm, 2025-01-02:
**Describe the bug**
In a loop I am generating a myDH_robot from some DH parameters, providing q_targets and calculating the forward kinematics and the jacobian of these q_targets. There are roughly 500,000 targets that are provided and then analyzed using the functions myDH_robot.fkine and myDH_robot.jacob0 the generated values are then stored. Afterwards the generated robot is deleted along with any arrays that have values stored to them. Furthermore gc.collect is run to ensure appropriate removal of items. 

This loop incrementally demands more memory at every iteration and by about iteration 16 the loop has demanded nearly 100% of the systems memory resulting in a crash.

After checked through my implementation for a potential leak I have determined by tracemalloc that the leak is to do with ET.py, ETS.py, and fknm.cpp, which to my understanding are responsible for the initialization of DH_robots. My understanding is that in my unique implementation where I am generating, analyzing, and then discarding the DH_robot results in the DH_robot being removed in the scope of my script. However it does not remove it from the scope of the toolbox. 

tracemalloc pointed to line 117 in ET.py and line 62 in ETS.py 

when looking through ET.py it appears that in my case the C objects reference is lost, but the object is not deleted which then results in the accumulation through the loop. This appears to track when following to the *ET_init PyObject in fknm.cpp where the pointer might be getting lost but not deleted 

 

**Version information**

Did you install from PyPI or GitHub?
installed from github on Commit 2a8111b

Robotics Toolbox depends heavily on two other packages: Swift (3D graphics) and SpatialMath toolbox (underpinning maths utilities).  If you think your issue is related to these, then please answer the questions above for them.

No issues related to swift or spatialmath toolbox

**To Reproduce**
Create a DH_robot class that pulls from an array of DH_parameters. Create a loop that generates unique parameters which is then passed to the generated class. From there using the functions of the class for a loop of q_targets.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

[image]

[image]

**Environment (please complete the following information):**
 - Ubuntu 22.04

**Additional context**
Let me know if there is anything I can help clarify

PCrnjak, 2025-06-14:
I am experiencing exact same problem you are describing in my application and its making my code unusable. I am changing ETS.py, ET.py, fknm.cpp to see where the exact problem. If you found any patch or fix that would be awesome

shubh-arm, 2025-08-13:
Hi PCrnjak, love your work at source. I ended up getting around the issue using forked processes where my computation was done via child processes then killed after the parent read the pickled object.

petercorke, 2026-09-19:
Thank you for your interest in RTB! The `fknm.cpp` extension you referenced was rewritten using nanobind since this was filed. I reproduced your scenario at scale (200 fresh `DHRobot` instances, ~400k `fkine`/`jacob0` calls) on current code and memory stays flat. Closing as fixed; thanks for the detailed report.

closed 2026-09-19 by petercorke