petercorke/robotics-toolbox-python #471: How to get all results of inverse kinematic computation?

Gian-Hacher, 2025-01-20:
For one pose of the end effector, there may be more than one feasible configuration of the robot arm, but the function 'ik_xx' can only compute one of them. How to get all possible results through these functions?

PhilFreeman, 2025-01-22:
The IK_xx functions are numerical methods to solve inverse kinematics. They converge on a single solution using gradient-decent-like methods. If you want to find "all" IK solutions (for certain manipulators and certain conditions there can be an infinite set of solutions) you can call the IK solver with multiple initial starting positions (q0) distributed across the configuration space. Each solver *may* find a solution. If there are a finite number of solutions, this will likely converge to that set of solutions. If there is an infinite set of solutions, the found solutions will lie on the self-motion manifold.

petercorke, 2026-08-25:
Thanks for the question — looks like this already got a solid answer in the thread above (sample the solver with multiple, distributed `q0` starting points to explore different local solutions, since the numerical IK methods converge on one solution at a time rather than enumerating all of them). Closing as answered — feel free to reopen or file a fresh issue if that doesn't cover what you needed.

closed 2026-08-25 by petercorke