petercorke/robotics-toolbox-python #474: Python 3.11 NOT INSTALLING / BUILDING

jakobbaumgartner, 2025-01-27:
It would appear that Python Robotics Toolbox version 1.10 does not work when installing on Python >= 3.11. I have tried installing it on windows 11 - Python  3.11.3 as well as on ubuntu 24.04 (WSL2) - Python 3.12.3, using github repo and pip. 

I was able to built version 1.10 on Ubuntu 22.04, which uses Python 3.10.12. 

```

Building wheels for collected packages: matplotlib
  Building wheel for matplotlib (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for matplotlib (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [858 lines of output]
      <string>:70: SetuptoolsDeprecationWarning: The test command is disabled and references to it are deprecated.
      !!

              ********************************************************************************
              Please remove any references to `setuptools.command.test` in all supported versions of the affected package.

              This deprecation is overdue, please update your project and remove deprecated
              calls to avoid build errors in the future.
              ********************************************************************************

      !!
      /tmp/pip-build-env-qs2vpavc/overlay/lib/python3.12/site-packages/setuptools/dist.py:701: SetuptoolsDeprecationWarning: The namespace_packages parameter is deprecated.
      !!

              ********************************************************************************
              Please replace its usage with implicit namespaces (PEP 420).

              See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages for details.
              ********************************************************************************

      !!
        ep.load()(self, ep.name, value)
      WARNING setuptools_scm.pyproject_reading toml section missing 'pyproject.toml does not contain a tool.setuptools_scm section'
      Traceback (most recent call last):
        File "/tmp/pip-build-env-qs2vpavc/normal/lib/python3.12/site-packages/setuptools_scm/_integration/pyproject_reading.py", line 36, in read_pyproject
          section = defn.get("tool", {})[tool_name]
                    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
      KeyError: 'setuptools_scm'
      /tmp/pip-build-env-qs2vpavc/normal/lib/python3.12/site-packages/setuptools_scm/git.py:312: UserWarning: git archive did not support describe output
        warnings.warn("git archive did not support describe output")

      Edit mplsetup.cfg to change the build options; suppress output with --quiet.

      BUILDING MATPLOTLIB
            python: yes [3.12.3 (main, Jan 17 2025, 18:03:48) [GCC 13.3.0]]
          platform: yes [linux]
             tests: no  [skipping due to configuration]
            macosx: no  [Mac OS-X only]

      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-x86_64-cpython-312

```

part 2 error msg:

```

z3xbqyoj/matplotlib_4fbb9e414dac44e9a2f07f9ea9bd22b9/build/freetype-2.6.1/objs/ftbzip2.o /tmp/pip-install-z3xbqyoj/matplotlib_4fbb9e414dac44e9a2f07f9ea9bd22b9/build/freetype-2.6.1/objs/psaux.o /tmp/pip-install-z3xbqyoj/matplotlib_4fbb9e414dac44e9a2f07f9ea9bd22b9/build/freetype-2.6.1/objs/psnames.o
      ar: `u' modifier ignored since `D' is the default (see `U')
      libtool: link: /usr/bin/gcc-ranlib /tmp/pip-install-z3xbqyoj/matplotlib_4fbb9e414dac44e9a2f07f9ea9bd22b9/build/freetype-2.6.1/objs/.libs/libfreetype.a
      libtool: link: ( cd "/tmp/pip-install-z3xbqyoj/matplotlib_4fbb9e414dac44e9a2f07f9ea9bd22b9/build/freetype-2.6.1/objs/.libs" && rm -f "libfreetype.la" && ln -s "../libfreetype.la" "libfreetype.la" )
      rm -f builds/unix/freetype-config builds/unix/freetype-config.tmp
      sed -e 's|%LIBSSTATIC_CONFIG%|-lfreetype|'   \
          -e 's|%build_libtool_libs%||' \
          -e 's|%exec_prefix%|/usr/local|'               \
          -e 's|%ft_version%|18.1.12|'                 \
          -e 's|%includedir%|/usr/local/include|'                 \
          -e 's|%libdir%|/usr/local/lib|'                         \
          -e 's|%prefix%|/usr/local|'                         \
          /tmp/pip-install-z3xbqyoj/matplotlib_4fbb9e414dac44e9a2f07f9ea9bd22b9/build/freetype-2.6.1/builds/unix/freetype-config.in                                                 \
          > builds/unix/freetype-config.tmp
      chmod +x builds/unix/freetype-config.tmp
      chmod a-w builds/unix/freetype-config.tmp
      mv builds/unix/freetype-config.tmp builds/unix/freetype-config
      rm -f builds/unix/freetype2.pc builds/unix/freetype2.pc.tmp
      sed -e 's|%REQUIRES_PRIVATE%||'     \
          -e 's|%LIBS_PRIVATE%||'             \
          -e 's|%build_libtool_libs%||' \
          -e 's|%exec_prefix%|/usr/local|'             \
          -e 's|%ft_version%|18.1.12|'                 \
          -e 's|%includedir%|/usr/local/include|'               \
          -e 's|%libdir%|/usr/local/lib|'                       \
          -e 's|%prefix%|/usr/local|'                       \
          /tmp/pip-install-z3xbqyoj/matplotlib_4fbb9e414dac44e9a2f07f9ea9bd22b9/build/freetype-2.6.1/builds/unix/freetype2.in                                                 \
          > builds/unix/freetype2.pc.tmp
      chmod a-w builds/unix/freetype2.pc.tmp
      mv builds/unix/freetype2.pc.tmp builds/unix/freetype2.pc
      Extracting /project/freetype/freetype2/2.6.1/freetype-2.6.1.tar.gz
      Building freetype in build/freetype-2.6.1
      error: Failed to download any of the following: ['http://www.qhull.org/download/qhull-2020-src-8.0.2.tgz'].  Please download one of these urls and extract it into 'build/' at the top-level of the source repository.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for matplotlib
Failed to build matplotlib
ERROR: Could not build wheels for matplotlib, which is required to install pyproject.toml-based projects`

```

raoshashank, 2025-01-30:
I was able to solve this by downloading qhull from this link: http://www.qhull.org/download/qhull-2020-src-8.0.2.tgz and extracting it to the build folder (as suggested in the error message).

jakobbaumgartner, 2025-01-30:
I was not successfull in this, however when using Ubuntu 22.04, which has older Python version installed, the toolbox works. Do wish they updated it.

closed 2026-08-25 by petercorke