petercorke/robotics-toolbox-python #481: Issues with teach ,

Rocket2503, 2025-02-14:
I am trying to use the teach function in my code but am getting this error in response. I cannot find where the 7 different arguments are given.

This is the error message I am recieving.
 
Message=Slider.__init__() takes 5 positional arguments but 7 were given

TypeError: Slider.__init__() takes 5 positional arguments but 7 were given

Nikolay-Zagrebin, 2025-03-08:
the same problem, an example from the book

e = ET2.R() * ET2.tx(1)
e.teach(0)
------------------------
Exception has occurred: TypeError
Slider.__init__() takes 5 positional arguments but 7 were given
  File "C:\work\Project\test1.py", line 41, in <module>
    e.teach(0)
TypeError: Slider.__init__() takes 5 positional arguments but 7 were given

Nikolay-Zagrebin, 2025-03-20:
> Same "TypeError: Slider.**init**() takes 5 positional arguments but 7 were given"
> 
> then,
> 
> I changed the inputs of slider() from positional argument to keyword argument, and solved my issue. the error targets to this location: \roboticstoolbox\backends\PyPlot\PyPlot.py, line 574.
> 
> By definition of Slider(SliderBase), only the first 4 parameters are positional, so the rest should be written as keyword argument, like： valinit= np.degrees(q[j]), valfmt= "% .1f°"
> 
> I hope it helps.

Thanks for the help!!!
To clarify a bit, the solution works if the robot is of the rtb.Drobot class, but if it is of the rtb.Robot2 class (as in the example above), then it needs to be corrected  \roboticstoolbox\backends\PyPlot\PyPlot2.py, line 346.

closed 2026-08-26 by petercorke