petercorke/robotics-toolbox-python #489: RNE method not working for ERobot

vkzimmerman, 2025-04-01:
**Describe the bug**
I am running into a bug where I am trying to use the rne() function on an inherited ERobot and am getting an error "TypeError: list indices must be integers or slices, not NoneType".

**Version information**
roboticstoolbox-python version 1.1.1
spatialmath-python version 1.1.13

**To Reproduce**
Steps to reproduce the behavior:
`robot2 = rtb.models.DH.Panda()`
`print(robot2.rne(q=np.array([-1.64904312,-0.10738241,0.0,0.,0.,0.,0.]),qd=np.array([0.,0.,0.,0.,0.,0.,0.]),qdd=np.array([0.,0.,0.,0.,0.,0.,0.]),gravity=np.array([9.81,0.,0.])))`
`robot3 = rtb.models.ETS.Panda()`
`print(robot3.rne(q=np.array([-1.64904312,-0.10738241,0.0,0.,0.,0.,0.]),qd=np.array([0.,0.,0.,0.,0.,0.,0.]),qdd=np.array([0.,0.,0.,0.,0.,0.,0.]),gravity=np.array([9.81,0.,0.])))`
`robot = rtb.models.URDF.Panda()`
`print(robot.rne(q=np.array([-1.64904312,-0.10738241,0.0,0.,0.,0.,0.]),qd=np.array([0.,0.,0.,0.,0.,0.,0.]),qdd=np.array([0.,0.,0.,0.,0.,0.,0.]),gravity=np.array([9.81,0.,0.])))`

**Expected behavior**
The expected behavior for the example ERobot rne call should be the same result as the rne call with the DH robot. The code runs an example DH robot, ETS robot, and a URDF Erobot. All three return different results. The ERobot especially causes a code breaking error message which is shown in the screenshot below.

**Screenshots**

[image]

**Environment (please complete the following information):**
 -WSL
-Python 3.10.12

**Additional context**
Add any other context about the problem here.

Crejan, 2025-09-02:
I also ran into this error using accel function on an Puma from URDF:

Here is a minimal example derived from the one in the documentation of accel
```
import numpy as np
import roboticstoolbox as rtb
#puma = rtb.models.DH.Puma560()  # <-- works fine
puma = rtb.models.URDF.Puma560()  # <-- throws an error
puma.accel(puma.qz, 0.5 * np.ones(6), np.zeros(6))
```

Traceback is the same:
```
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    puma.accel(puma.qz, 0.5 * np.ones(6), np.zeros(6))
  File "roboticstoolbox/robot/Dynamics.py", line 500, in accel
    M = self.rne(qI, qdI, qddI, gravity=[0, 0, 0])
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "roboticstoolbox/robot/Robot.py", line 1794, in rne
    v[j] = Xup[j] * v[jp] + vJ
                    ~^^^^
  File "spatialmath/spatialvector.py", line 148, in __getitem__
    return self.__class__(self.data[i])
                          ~~~~~~~~~^^^
TypeError: list indices must be integers or slices, not NoneType
```
-> Debugger shows that:
- jp is None. which is jindex of parent of link[1]
  - parent of link[1] is link[0]
  - jindex on link[0] is None

Here:
https://github.com/petercorke/robotics-toolbox-python/blob/2a8111b10dd09c5ce8b8a94495c84ff27bcb6862/roboticstoolbox/robot/Robot.py#L1789
and here:
https://github.com/petercorke/robotics-toolbox-python/blob/2a8111b10dd09c5ce8b8a94495c84ff27bcb6862/roboticstoolbox/robot/Robot.py#L1806
is a check if the link parent is None but there is no check if parent.jindex is None. I guess the problem is the additional root link without joint in the urdf?

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking close to a year and a half to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one, I owe you a more honest answer than "fixed, please retry": I reproduced your exact repro on current `main` — the `TypeError` is gone, but the URDF Panda's `rne()` now silently returns `[0,0,0,0,0,0,0]`, which is wrong (the DH Panda, same pose, returns real nonzero torques). Same underlying root cause as #236/#463, tracked properly as #636, which references a real working patch already proposed in #483. Thank you for the report.

closed 2026-08-25 by petercorke