petercorke/robotics-toolbox-python #496: ERROR: Could not build wheels for roboticstoolbox-python, which is required to install pyproject.toml-based projects

QNjeason, 2025-06-20:
PS C:\Users\hxy> python --version
Python 3.9.13

install report error

      "D:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.44.35207\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -I./roboticstoolbox/core/ "-ID:\Program Files\Python39\include" "-ID:\Program Files\Python39\Include" "-ID:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.44.35207\include" "-ID:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\VS\include" /Tc./roboticstoolbox/core/frne.c /Fobuild\temp.win-amd64-cpython-39\Release\roboticstoolbox\core\frne.obj
      frne.c
      ./roboticstoolbox/core/frne.c(28): fatal error C1083: 无法打开包括文件: “math.h”: No such file or directory
      error: command 'D:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.44.35207\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for roboticstoolbox-python
Failed to build roboticstoolbox-python
ERROR: Could not build wheels for roboticstoolbox-python, which is required to install pyproject.toml-based projects

cumtlqf, 2025-10-04:
met same problem

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking just over a year to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one: at the time, `pip install` had to compile RTB's C extensions locally with MSVC, which is where this broke. RTB now ships prebuilt wheels (via cibuildwheel) for Windows, so a normal install shouldn't need a local compiler anymore. One important note though: your report was against Python 3.9, and RTB now requires Python 3.10 or newer — so if you're still on 3.9, you'd need to upgrade Python first regardless of the wheel situation.

While I can't promise every corner of this has been exercised, I'd like to ask you to retry it on a supported Python version (if you're still motivated) and get back to me with a comment here, or a new issue, if the problem remains.

closed 2026-08-25 by petercorke